PhilipsHue/flutter_reactive_ble #467: [Android]

VinniciusJesus, 2021-12-08:
**Cellphone 1: Iphone 7 (IOS 15.1)**
**Cellphone 2: Redmi Note 9 Pro Xiaomi (Android 11)**
**Cellphone 3: A10 Samsung (Android 10)**

**Cellphone 1** and **Cellphone 2** is work just fine!

Cellphone 3 When connecting, it is disconnected right away. I don't know if there's anything the package can do, but I'm sharing this information in case other people are having the same problem.

All tests were with the same code.

Flutter: 2.5.3
Flutter_Reactive_ble: 5.0.2

Taym95, 2021-12-13:
I will try to test it and get back to you, but I am afraid it is a device-specific issue!

VinniciusJesus, 2021-12-13:
Other installed apps were trying to connect at the same time on **Cellphone 3**

closed 2021-12-13 by VinniciusJesus