PhilipsHue/flutter_reactive_ble #468: Do not receive DeviceConnectionState.connecting on iOS example app v 5.0.2

cjcdev, 2021-12-08:
**Describe the bug**
On iOS, I do not receive DeviceConnectionState.connecting state before the Connected state.

On Android, I am always seeing Connecting state.

**To Reproduce**
Steps to reproduce the behavior:
1. Use example code app from 5.0.2 package.
2. Connect to device
3. Disconnect from device
4. Look at the logs.  You will see a Connected event, but not a Connecting event.

**Expected behavior**
I should always see Connecting state prior to Connected state

**Smartphone / tablet**
 - Device: iPhone 7 Plus, Samsung Galaxy Tab A7 
 - OS: iOS 15.1, Android: 11.0
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: Cypress BLE based device
- Does it run a custom firmware / software: yes

arthurv, 2022-04-17:
Can confirm this issue as well - we can see a connected event, but DeviceConnectionState.connecting never happens.

pzehle, 2022-05-02:
Any updates on this?

I am also having this issue. This comes as a big problem when the user decides to click on "Cancel" when the pair prompt appears. Because connecting never happens, the library thinks the user clicked on "Pair", which is not always true.

Anyway, when the prompt appears, library reports as connected, so there is actually no way to know for sure if the device is actually connected or not. Only way for me to do this right now is to attempt the connection 2 times.

Any better ideas?

arthurv, 2022-05-02:
This issue has actually been mentioned elsewhere: https://github.com/PhilipsHue/flutter_reactive_ble/issues/385#issuecomment-1040502145

And I can also confirm the linked comment - the "connecting" state isn't communicated on the first connect, but if I disconnect and reconnect, the "Connecting" state is seen.

santa010, 2022-05-10:
Any news on the issue?

santa010, 2022-05-24:
Did anyone find any better workaround for the 1st connection issue, or is disconnect and then reconnect the way to go for now?

mckn, 2022-07-24:
Same issue here, any news regarding this?

trueb2, 2022-11-04:
> This issue has actually been mentioned elsewhere: [#385 (comment)](https://github.com/PhilipsHue/flutter_reactive_ble/issues/385#issuecomment-1040502145)
> 
> And I can also confirm the linked comment - the "connecting" state isn't communicated on the first connect, but if I disconnect and reconnect, the "Connecting" state is seen.

#385 is not resolved, but the issue hasn't been re-opened. The fix was reverted in https://github.com/PhilipsHue/flutter_reactive_ble/pull/440 according to https://github.com/PhilipsHue/flutter_reactive_ble/issues/385#issuecomment-1040524572. The error `Warning! No event channel set up to report a connection update` is still happening on iOS in 5.0.2.

Attempting but failing to mitigate the issue, I do several steps recommended in various issues
1. `WidgetsFlutterBinding.ensureInitialized();` as the first line in main
2. Initialize a `FlutterReactiveBle ble` instance as soon as I have permission approved
3. Await a `ble.initialize()` call
4. Await 100 milliseconds
5. Subscribe to ble.statusStream
  1. If ble.status = BleStatus.ready, then cancel existing scan subscription, await 50 milliseconds, and start a new one.
6. before connectTo on a DiscoveredDevice, await 100 milliseconds
7. before subscribing to characteristics after connect, await 500 milliseconds

This sequence does not mitigate the issue but seems reliable for everything else.

vanlooverenkoen, 2023-01-19:
I'm having the same issue on iOS using 5.0.3