PhilipsHue/flutter_reactive_ble #469: cancelling the streaming sometimes doesn't disconnect

lph65724, 2021-12-09:
I have a page that reads information from a ESP32 device. When popping the page, connectedDeviceStream.cancel() is called to disconnect the device, but it sometimes doesn't disconnect.  This doesn't happen often, but it happens.  When it happens, the device doesn't seem to receive the disconnect command, so the device thinks it's still connected, so it cannot be reconnected again unless I close the app.  Is there a way I can know if the device is actually disconnected before leaving the page?  Is it a good idea to call cancel() a few more times to ensure the device is disconnected?  Thank you

Taym95, 2021-12-13:
Can you test with an example app? canceling the stream should disconnect the device!

remonh87, 2021-12-13:
Please also read this blog post: https://devzone.nordicsemi.com/f/nordic-q-a/38422/android-disconnect-and-close-do-not-disconnect .  Unfortunately Android is known for being unreliable in disconnecting so I think this explains your flaky behaviour sometimes.

lph65724, 2021-12-14:
> Can you test with an example app? canceling the stream should disconnect the device!

Canceling the stream does disconnect the device, but it sometimes fails to disconnect.

lph65724, 2021-12-14:
> Please also read this blog post: https://devzone.nordicsemi.com/f/nordic-q-a/38422/android-disconnect-and-close-do-not-disconnect . Unfortunately Android is known for being unreliable in disconnecting so I think this explains your flaky behaviour sometimes.

Ok, it explains why canceling the stream sometimes fails to disconnect the device.

closed 2021-12-14 by lph65724