PhilipsHue/flutter_reactive_ble #470: Can we have a Disconnect method , without cancelling the Stream

rohitsangwan01, 2021-12-11:
Hey , Cancelling Stream to Disconnet works really Great , but i want my app to run in background and using https://pub.dev/packages/flutter_background , this package to keep app active in background ,, works really great but now , after using foreground services , when i swipe up app from memory , bluetooth device still remains connected , and strangely VsCode also show app still connected in Debug mod , but in my Mobile , its actually terminated , seems like app isolates still active , but then issue is Stream is not cancelling somehow , and device remains connected until i toggle Mobile Bluetooth , So either i can save Device ID locally , so that next time when app open Pass that Id to Disconnect method before scan , something like that , or maybe some way to kill stream when swipe up app 
Device One Plus 8T
Os Android 11

Taym95, 2021-12-13:
>>> but then issue is Stream is not cancelling somehow

Can't you write a code that cancels the stream in `dispose` or when the app is killed?

rohitsangwan01, 2021-12-13:
Yeah, Thanks for reply
In case if Someone else in Same situation , what i did is ,set a service terminate callback from native Android to Flutter ,which gets executed when user swipe up app, and Cancelling all streams with all foreground services

Taym95, 2021-12-13:
Can you just use [this](https://stackoverflow.com/questions/52074265/flutter-detect-killing-off-the-app) from flutter?

rohitsangwan01, 2021-12-13:
Lyfecycle.detach was not calling when i tried. , By using a Library which Starts foreground services , When i swipe up App , it still shows App connected to Vscode ( usually we get  ," Lost connection from device " ) , Evenr Hot reload and Restart works

Taym95, 2021-12-13:
@remonh87 do you need a disconnect method? IMHO I don't see the need for it since the logic can be implemented inside the app!

rohitsangwan01, 2021-12-13:
a simple method such as , tryDisconnect(address) will be really helpfull , if its not taking much of your time , otherwise cancel stream is also good option

remonh87, 2021-12-13:
The problem is that a disconnect method is simple to implement but it goes:
1. against the nature of our library where we manage the lifecycle through streams. Because we know the exact state we limit the amount of GATT 133 errors which happen a lot when you have free disconnect methods. I am aware it comes at a cost
2. Please check my response on https://github.com/PhilipsHue/flutter_reactive_ble/issues/322#issuecomment-989102539 .

closed 2021-12-13 by remonh87