PhilipsHue/flutter_reactive_ble #471: [iOS] Add option to set CBCentralManagerOptionShowPowerAlertKey

JMNewsome, 2021-12-11:
Can an option during initialization be added to control the CBCentralManagerOptionShowPowerAlertKey

` self.centralManager = CBCentralManager(
            delegate: centralManagerDelegate,
            queue: nil,
            //NOTE - This is to prevent iOS popup alerting the user to open up settings
            options: [CBCentralManagerOptionShowPowerAlertKey: false]
        )`

This should make it easier to allow if the app should alert the user or not when the bluetooth setting is turned off. Currently I have set this false in my local version of the code in swift (Central.swift) but controlling this from flutter would be cleaner.

remonh87, 2021-12-15:
@werediver  can you look into this request?

werediver, 2021-12-18:
This is sort of a reasonable request, but the downside of disabling the BLE power alert is that it won't be possible to trigger it later (not without reinitializing `CBCentralManager`, which we don't support [I think]), which means the app would need to instruct its user on how to enable Bluetooth, if/when needed.

An alternative is to not initialize the plug-in, unless the app is positive it needs to use BLE. In that case the power alert will only appear, when the app needs BLE to be enabled, and that will be the most straightforward way to do it for the user.