PhilipsHue/flutter_reactive_ble #475: Reading "Generic Access Service (UUID: 1800)" Throwing " Generic Access Profile is not found " Error in IOS only

rohitsangwan01, 2021-12-15:
**Describe the bug**
When am Trying yo read Generic Access Service (UUID: 0x1800) , am getting Error on IOS
```dart
GenericFailure<CharacteristicValueUpdateError>(code: CharacteristicValueUpdateError.unknown, message: "A service Generic Access Profile is not found in the peripheral 6D8D8DC9-BA02-430F-4DC7-572E057D801B (make sure it has been discovered)")
```
Altough its working Perfectly fine on Android with Same Code ,Tried to print "DiscoveredService List " on both platforms and results are ,

On IOS ->
flutter: 180a
flutter: 81f12000-59c5-4255-bad2-7685cc587fd3
flutter: fe59
flutter: 180f

on Android ->
flutter : 00001800-0000-1000-8000-00805f9b34fb
flutter : 00001801-0000-1000-8000-00805f9b34fb
flutter : 0000180a-0000-1000-8000-00805f9b34fb
flutter : 81f12000-59c5-4255-bad2-7685cc587fd3
flutter : 0000fe59-0000-1000-8000-00805f9b34fb
flutter : 0000180f-0000-1000-8000-00805f9b34fb

**Smartphone / tablet**
 - Device: iPhone 12 Pro / One Plus 8T 
 - OS:  iOS 15 / Android 11
 - Package version: 5.0.2

**Peripheral device**
- Nrf

rohitsangwan01, 2021-12-17:
@remonh87 , tried with Flutter_Blue package as well , but same output , not sure why Ios is not discovering that service with UUID (1800) , here is the result from Flutter_blue , seems like issue with device , not library

```dart
Service  : 0000180a-0000-1000-8000-00805f9b34fb
Service  : 81f12000-59c5-4255-bad2-7685cc587fd3
Service  : 0000fe59-0000-1000-8000-00805f9b34fb
Service  : 0000180f-0000-1000-8000-00805f9b34fb
```

Taym95, 2021-12-17:
Hi, it would be really great to follow the issue template and double check before creating issue(check on example app, check on Rnf connect app...), a question in stack overflow would also be good in this case, just make diff between library issues and device/app specific issue.

rohitsangwan01, 2021-12-18:
@Taym95 , i think its a ios related issue not library , i was hoping some workaround for this , but seems like we can't do it in IOS , so i created another characterstic in device to rename  ,which worked for me , Thanks for response ,i guess we can close this issue now

closed 2021-12-18 by rohitsangwan01