PhilipsHue/flutter_reactive_ble #478: Support for multiple characteristics with the same Uuid

re-mike, 2021-12-21:
Some BLE Profiles allow for having multiple characteristics within a service that have the same Uuid and are uniquely identified by the characteristic descriptor (e.g. Automation IO).

Is it possible to support multiple characteristics with the same Uuid within a single service?

For example, if the List<DiscoveredCharacteristic> in the DiscoveredService class had an additional unique field so that the list could contain multiple characteristics with the same Uuid.

I understand that the second 'u' in 'Uuid' already implies that this should uniquely identify the characteristic :)
But the Automation IO Service specification states:

_"If a device has more than one instance of the Analog characteristic, each characteristic shall include a Characteristic Presentation Format descriptor that has a namespace / description value that is unique for that instance of the Analog characteristic... Description values from 0x0001 and upwards shall be used to uniquely identify each Analog characteristic."_

[https://www.bluetooth.com/specifications/specs/automation-io-service-1-0/](url)

remonh87, 2022-01-11:
@re-mike Sorry for the late reaction (holidays etc).  I have no clue to be honest our library doesn't do anything to prevent this from happening but I am not sure if the Android and/or iOS ble implementation does something against it. I am very curious to learn about your findings :)

spkersten, 2022-11-07:
I'll be picking up this feature soon.

spkersten, 2022-11-16:
On both Android and iOS, the BLE API is such that you first have to discover services and characteristics and then use the found objects. Our library currently makes the assumption that there is a unique characteristic of a given type and simply selects the first with a given type id.

I’m proposing to change (or at least amend) our interface to match that. Concretely the following:

Extend `DiscoveredCharacteristic` with two fields to uniquely identify a characteristic: `int serviceInstanceId` and `int characteristicInstanceId`. Both fields have limited validity and should not be stored. (They could change between restarts/reconnects of the peripheral.) Make it non-constructible for users, to avoid misuse. (Private constructors, runtime check that we don’t get a subclass.) On Android `characteristicInstanceId` is enough to uniquely identify a characteristic and `serviceInstanceId` will be unused (set to 0), as the platform characteristic object exposes such an id. On iOS, there is nothing on the `CBCharacteristic` interface to keep two of the same type apart. So both ids will be, respectively, the index in the list of `CBService`s of the same type, and the index in the list of `CBCharacteristics` of the same type within a `CBService`. (If this proves unsuitable, we’d need to keep a dictionary of unique ids to `CBCharacteristic` objects).

Next to existing `discoverServices`, add `getDiscoveredServices` to make discovered services/chars available on iOS. On Android it will forward to `discoverServices` (which uses a cache) as there is no getter of services exposed by RxAndroidBle and the OS doesn’t allow discovery of individual services anyway. 

Add to `FlutterReactiveBle`: 
- `read(DiscoveredCharacteristic)`, 
- `writeWith(out)Response(DiscoveredCharacteristic)`,
- `subscribeTo(DiscoveredCharacteristic)`

Update existing `readCharacteristic`/`writeCharacteristicWith(out)Response`/`subscribeToCharacteristic` to use (in Dart) `getDiscoverServices` to get the first matching characteristic and use the new `read`/`write`/`subscribeTo` methods. These methods are kept for backwards compatibility. Maybe these can be deprecated later or kept as shortcuts.

This would also fix #509.

curiousmuch, 2023-02-12:
Hello! 

I believe issue #509 should be closed as it's a duplicate of this issue. I'd be willing to work on this, but I'd like confirmation the team is willing to merge the pull request upstream once it meets your coding standard.

PR #776 (2023-08-28): Support multiple services or characteristics with the same id
Fixes #478
Fixes #509

The change is mostly backwards compatible. The exception to that is that if you try to use the existing read/write/subscribe methods for a characteristic ids for which there is no single unique instance, an exception is thrown. Before, it would read/write/subscribe the first matching characteristic.
closed 2023-08-28 by Taym95