PhilipsHue/flutter_reactive_ble #481: loss data

oooldCaption, 2021-12-24:
**Describe the bug**
I have a subscription with two types of data, one with a frequency of 100hz and one with a frequency of 50hz. But in 20 seconds, I only received 150~500 sets of data, which should be around 750 data.

I use nrf connect to view the log,  and the data in 5s seconds is 740~750;

**To Reproduce**
Steps to reproduce the behavior:
0. ScanForDevices
1. Connect to one `connectTo()`
2. SubscribeToCharacteristic
3. writeCharacteristicWithoutResponse

**Expected behavior**
A clear and concise description of what you expected to happen.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: [e.g. iPhone X]
 - OS: [e.g. iOS 13, Android 10]
 - Package version: [e.g. 1.0.0]

**Peripheral device**
- Vendor, model: [e.g. Fitbit Versa, iPhone SE, or CUSTOM]
- Does it run a custom firmware / software: yes / no

**Additional context**
Add any other context about the problem here.

MrCsabaToth, 2021-12-25:
So is this iOS only or Android? What's the Android 10 device? I hope it's not an old slow Android tablet from a bottom of a drawer because those old devices can indeed not be able to handle 50Hz data update!

oooldCaption, 2021-12-25:
> 那么这是iOS还是Android？什么是安卓10设备？我希望它不是一个旧的缓慢的Android平板电脑从抽屉的底部，因为那些旧设备确实无法处理50Hz的数据更新！

soory; Some information is not saved,I updated the content of the issue
Android 11, VIVO IQCC neo3

In **lutter_reactive_ble**  lib, I tested ios and android platforms and it happens.
In **flutter_blue** lib, it's only happens on the Android platform.

MrCsabaToth, 2021-12-25:
Thanks for your efforts. Do you see any error messages related to this in the LogCat?

oooldCaption, 2021-12-25:
> Thanks for your efforts. Do you see any error messages related to this in the LogCat?
> 
> 感谢您的努力。您在 LogCat 中看到任何与此相关的错误消息了吗？
Good morning, greetings from Shanghai. hhhh;
 No...,  on the console I only saw what I printed with `print()`, I will check the output in logcat later.

oooldCaption, 2021-12-25:
> 
logcat output :

```
2021-12-25 09:10:05.779 10646-10781/com.example.flutter_demo I/flutter: ********[3, 0, 0, 2, 8, 0, 0, 3, 3, 231, 11, 0, 0, 251, 7, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
...
Duplicate line 2 ~ line159 
...
2021-12-25 09:10:07.517 10646-11387/com.example.flutter_demo D/BluetoothGatt: onPhyUpdate() - status=0 address=DE:7F:B0:4F:65:13 txPhy=2 rxPhy=2
2021-12-25 09:10:07.520 10646-10781/com.example.flutter_demo I/flutter: ********[3, 0, 0, 2, 8, 0, 0, 3, 3, 218, 93, 1, 0, 178, 66, 1, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
...
Duplicate line 162 ~ line561
...
2021-12-25 09:10:10.581 10646-10781/com.example.flutter_demo I/flutter: ********[3, 0, 0, 2, 8, 0, 0, 4, 2, 9, 0, 255, 255, 254, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]

```
[image]

MrCsabaToth, 2021-12-25:
I wonder what happens with the data which is not arriving to your application layer. Is it simply thrown away, basically it happens as a built-in rate limit which is based on the handheld device's speed and processing capabilities, or do the packages pile up somewhere?

oooldCaption, 2021-12-27:
> I wonder what happens with the data which is not arriving to your application layer. Is it simply thrown away, basically it happens as a built-in rate limit which is based on the handheld device's speed and processing capabilities, or do the packages pile up somewhere?
> 

Just discard the data. The app does not tell the Bluetooth device how much data the app receives;  Is the handheld device a Bluetooth device?

remonh87, 2021-12-30:
hmm interesting issue not sure what I can do on the software to improve this.  Can you try to negotiate a lower MTU (e.g. 128) and what can help on Android is requesting a higher connection priority.

closed 2022-01-24 by oooldCaption