PhilipsHue/flutter_reactive_ble #482: android autoconnect doesn't work,  why?

compvisiondeveloper, 2021-12-24:
**Is your feature request related to a problem? (please describe)**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

remonh87, 2021-12-24:
you can use auto connect when you do not specify a timeout. See: https://github.com/PhilipsHue/flutter_reactive_ble/blob/9bd6ed07672c88ea99e9e4740012f8ebc4d02f2c/packages/flutter_reactive_ble/lib/src/reactive_ble.dart#L252

closed 2021-12-24 by remonh87

compvisiondeveloper, 2021-12-24:
Thanks for reply.
Actually i tried without a timeout but didn't work.Can you recommend any one else solution for this ?