PhilipsHue/flutter_reactive_ble #483: Accessing Raw Data

sonelektrikci, 2022-01-10:
**Is your feature request related to a problem? (please describe)**
We need a way to access rawdata from advertisement packages.

**Describe the solution you'd like**
Returning a rawdata would be very helpful for us. 

**Describe alternatives you've considered**
I checked the android documentation and saw a way to access rawdata. So it can be implemented to this plugin too.

remonh87, 2022-01-11:
Indeed Android has the following method: 
https://developer.android.com/reference/android/bluetooth/le/ScanRecord#getBytes()

The problem is that I do not see a way of retrieving the raw advertisement data on iOS. I only saw Manufacturerdata and servicedata which we expose in this lib. @werediver  can you confirm this?

sonelektrikci, 2022-01-11:
> f retrieving the raw advertisement data on iOS.

We can't access it with the flutter_reactive_ble tho right? I have to change the kotlin part of the plugin.

werediver, 2022-01-12:
Only the data mentioned in [Advertisement Data Retrieval Keys](https://developer.apple.com/documentation/corebluetooth/cbcentralmanagerdelegate/advertisement_data_retrieval_keys) is accessible on iOS.

Some of that data is already accessible via "flutter_reactive_ble", see [DiscoveredDevice](https://github.com/PhilipsHue/flutter_reactive_ble/blob/9bd6ed07672c88ea99e9e4740012f8ebc4d02f2c/packages/reactive_ble_platform_interface/lib/src/model/discovered_device.dart#L29).

It's best to keep the library behavior the same on all the supported platforms and any asymmetry would need a good reason to be introduced.

sonelektrikci, 2022-01-13:
> Advertisement Data Retrieval Keys

I can't access to first link you provide. I thought CoreBluetooth lets us access the raw data, when I checked the documentation. 
Also for Android part, changing some codes in RxBleAndroid package can solve our problem. I'm not sure it's the right package tho.

werediver, 2022-01-13:
> I can't access to first link you provide.

Strangely, I couldn't read that page until I disabled "uBlock Origin". It's really weird, but 🤷

closed 2022-01-20 by sonelektrikci