PhilipsHue/flutter_reactive_ble #484: Scanning doesn't stop when StreamSubscription.cancel() is called

jobfeikens, 2022-01-10:
Method scanForDevices in reactive_ble_mobile_platform.dart is currently implemented as follows:

```
@override
  Stream<void> scanForDevices({
    required List<Uuid> withServices,
    required ScanMode scanMode,
    required bool requireLocationServicesEnabled,
  }) =>
      _bleMethodChannel
          .invokeMethod<void>(
            "scanForDevices",
            _argsToProtobufConverter
                .createScanForDevicesRequest(
                  withServices: withServices,
                  scanMode: scanMode,
                  requireLocationServicesEnabled:
                      requireLocationServicesEnabled,
                )
                .writeToBuffer(),
          )
          .asStream();
```

In this method, a Future\<void\> is converted to Stream\<void\> but nothing is done when that stream is cancelled as you can't cancel a Future in dart.

Am I missing something?

remonh87, 2022-01-11:
The magic is in happening here:

https://github.com/PhilipsHue/flutter_reactive_ble/blob/9bd6ed07672c88ea99e9e4740012f8ebc4d02f2c/packages/flutter_reactive_ble/lib/src/device_scanner.dart#L45 .

So we first invoke "scanForDevices" which creates under the hood an event channel that propagates scan results. When the stream is cancelled we also will close the event channel for the scan results which stops the ble scan. 

Hope this answers your question.

closed 2022-01-11 by remonh87

jobfeikens, 2022-01-24:
The ble scan is indeed stopped, however, the Stream returned by `scanForDevices` never completes. I made an example demonstrating the issue:
```
import 'dart:async';

import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';
import 'package:permission_handler/permission_handler.dart';

void main() {
  runApp(MyApp());
}

class MyApp extends StatefulWidget {

  @override
  State<StatefulWidget> createState() {
    return MyAppState();
  }
}

class MyAppState extends State<MyApp> {

  StreamSubscription? scanSubscription;

  bool isScanning = false;

  void initState() {
    super.initState();
    Permission.bluetoothScan.request();
    Permission.bluetoothConnect.request();
  }

  Future<void> start() async {

    setState(() {
      scanSubscription = FlutterReactiveBle()
          .scanForDevices(withServices: [])
          .listen((event) {});
    });
    try {
      await scanSubscription!.asFuture();
      /// This future never completes!!!

      print('Scan completed');

    } finally {
      setState(() => scanSubscription = null);
    }
  }

  Future<void> cancel() async {
    print('Cancelling $scanSubscription');
    await scanSubscription!.cancel();

    print('Cancelled');
  }

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      home: Scaffold(
        body: Column(
          mainAxisAlignment: MainAxisAlignment.center,
          crossAxisAlignment: CrossAxisAlignment.center,
          children: [
            ElevatedButton(
                onPressed: scanSubscription == null ? () => start() : null,
                child: Text('Start')
            ),
            ElevatedButton(
                onPressed: scanSubscription != null ? () => cancel() : null,
                child: Text('Cancel')
            )
          ],
        ),
      ),
    );
  }

  @override
  void dispose() {
    scanSubscription?.cancel();
    super.dispose();
  }
}

```

jobfeikens, 2022-01-26:
The future in the example above never completes because the scan channel never emits a "done" event (I think because it's a broadcast stream).

To make the above example work, I had to change the code a little:
<details>
  <summary>Code</summary>

  ```
  import 'dart:async';

import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';
import 'package:permission_handler/permission_handler.dart';
import 'package:rxdart/rxdart.dart';

void main() {
  runApp(MyApp());
}

class MyApp extends StatefulWidget {

  @override
  State<StatefulWidget> createState() {
    return MyAppState();
  }
}

class MyAppState extends State<MyApp> {

  bool? isScanning;
  StreamSubscription? scanSubscription;

  void initState() {
    super.initState();
    Permission.bluetoothScan.request();
    Permission.bluetoothConnect.request();
  }

  Future<void> start() async {
    setState(() {
      isScanning = true;

      scanSubscription = FlutterReactiveBle()
          .scanForDevices(withServices: [])
          .doOnCancel(() => setState(() => isScanning = false))
          .listen(print, cancelOnError: true);
    });
  }

  Future<void> cancel() async {
    if (isScanning ?? false) {
      await scanSubscription!.cancel();
    }
  }

  @override
  Widget build(BuildContext context) {
    return MaterialApp(
      home: Scaffold(
        body: Column(
          mainAxisAlignment: MainAxisAlignment.center,
          crossAxisAlignment: CrossAxisAlignment.center,
          children: [
            ElevatedButton(
                onPressed: !(isScanning ?? false) ? () => start() : null,
                child: Text('Start')
            ),
            ElevatedButton(
                onPressed: (isScanning ?? false) ? () => cancel() : null,
                child: Text('Cancel')
            )
          ],
        ),
      ),
    );
  }

  @override
  void dispose() {
    scanSubscription?.cancel();
    super.dispose();
  }
}

  ```
</details>