PhilipsHue/flutter_reactive_ble #486: Can not compile for IOS

ride4sun, 2022-01-10:
**Describe the bug**
I get error messages when compile for IOS saying I need IOS 10.0 or newer. I did read the comments on the other defect related to this but still can not solve this and I am stuck.
```
Xcode's output:
↳
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/location-4.3.0/ios/Classes/LocationPlugin.m:176:13: warning: 'UIAlertView' is deprecated: first deprecated in iOS 9.0 -
    UIAlertView is deprecated. Use UIAlertController with a preferredStyle of UIAlertControllerStyleAlert instead [-Wdeprecated-declarations]
                UIAlertView *alert = [[UIAlertView alloc] initWithTitle:@"Location is Disabled"
                ^
    In module 'UIKit' imported from /Users/jay/Desktop/Projects/obdchk/ios/Pods/Target Support Files/location/location-prefix.pch:2:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS15.2.sdk/System/Library/Frameworks/UIKit.framework/Headers/UIAlertView.h:27:12: note:
    'UIAlertView' has been explicitly marked deprecated here
    @interface UIAlertView : UIView
               ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/location-4.3.0/ios/Classes/LocationPlugin.m:176:48: warning: 'UIAlertView' is deprecated: first deprecated in iOS 9.0 -
    UIAlertView is deprecated. Use UIAlertController with a preferredStyle of UIAlertControllerStyleAlert instead [-Wdeprecated-declarations]
                UIAlertView *alert = [[UIAlertView alloc] initWithTitle:@"Location is Disabled"
                                                   ^
    In module 'UIKit' imported from /Users/jay/Desktop/Projects/obdchk/ios/Pods/Target Support Files/location/location-prefix.pch:2:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS15.2.sdk/System/Library/Frameworks/UIKit.framework/Headers/UIAlertView.h:27:12: note:
    'UIAlertView' has been explicitly marked deprecated here
    @interface UIAlertView : UIView
               ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/location-4.3.0/ios/Classes/LocationPlugin.m:303:57: warning: 'speedAccuracy' is only available on iOS 10.0 or newer
    [-Wunguarded-availability]
            @"speed_accuracy": superiorToIos10 ? @(location.speedAccuracy) : @0.0,
                                                            ^~~~~~~~~~~~~
    In module 'CoreLocation' imported from /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/location-4.3.0/ios/Classes/LocationPlugin.m:4:
    /Applications/Xcode.app/Contents/Developer/Platforms/iPhoneOS.platform/Developer/SDKs/iPhoneOS15.2.sdk/System/Library/Frameworks/CoreLocation.framework/Headers/CLLocation.h:393:56: note:
    'speedAccuracy' has been marked as being introduced in iOS 10.0 here, but the deployment target is iOS 9.0.0
    @property(readonly, nonatomic) CLLocationSpeedAccuracy speedAccuracy API_AVAILABLE(macos(10.15), ios(10.0), watchos(3.0), tvos(10.0));
                                                           ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/location-4.3.0/ios/Classes/LocationPlugin.m:303:57: note: enclose 'speedAccuracy' in an @available check to silence this
    warning
            @"speed_accuracy": superiorToIos10 ? @(location.speedAccuracy) : @0.0,
                                                            ^~~~~~~~~~~~~
    3 warnings generated.
    /Users/jay/Desktop/Projects/obdchk/ios/Pods/Reachability/Reachability.m:465:63: warning: cast to smaller integer type 'unsigned int' from 'Reachability *' [-Wpointer-to-int-cast]
                                 NSStringFromClass([self class]), (unsigned int) self, [self currentReachabilityFlags]];
                                                                  ^~~~~~~~~~~~~~~~~~~
    1 warning generated.
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData extras/BLEStatus.swift:3:29: error: 'CBManagerState' is only available in iOS
    10.0 or newer
    func encode(_ centralState: CBManagerState) -> Int32 {
                                ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData extras/BLEStatus.swift:3:6: note: add @available attribute to enclosing
    global function
    func encode(_ centralState: CBManagerState) -> Int32 {
         ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:14:46: error: 'CBManagerState' is only available in iOS
    10.0 or newer
        typealias StateChangeHandler = (Central, CBManagerState) -> Void
                                                 ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:14:15: note: add @available attribute to enclosing type
    alias
        typealias StateChangeHandler = (Central, CBManagerState) -> Void
                  ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:12:13: note: add @available attribute to enclosing class
    final class Central {
                ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:106:16: error: 'CBManagerState' is only available in iOS
    10.0 or newer
        var state: CBManagerState { return centralManager.state }
                   ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:106:9: note: add @available attribute to enclosing property
        var state: CBManagerState { return centralManager.state }
            ^
    /Users/jay/development/flutter/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:12:13: note: add @available attribute to enclosing class
    final class Central {
                ^
[... 604 lines cut ...]
    note: Using new build system
    note: Planning
    note: Build preparation complete
    warning: Multiple targets match implicit dependency for linker flags '-framework Reachability'. Consider adding an explicit dependency on the intended target to resolve this ambiguity. (in
    target 'Runner' from project 'Runner')
        note: Target 'Reachability' (in project 'Pods')
        note: Target 'ReachabilitySwift' (in project 'Pods')
    note: Building targets in dependency order
```

versions:
```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 2.8.1, on macOS 11.6 20G165 darwin-x64, locale en-US)
[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
[✓] Xcode - develop for iOS and macOS (Xcode 13.2.1)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2020.3)
[✓] VS Code (version 1.60.2)
[✓] VS Code (version 1.44.2)
[✓] VS Code (version 1.42.1)
[✓] VS Code (version 1.41.1)
[✓] Connected device (2 available)

```

remonh87, 2022-01-11:
I advice to check the `IPHONEOS_DEPLOYMENT_TARGET ` in your `project.pbxproj` file we need a minimum iOS version of 10. 

See for example: https://github.com/PhilipsHue/flutter_reactive_ble/blob/9bd6ed07672c88ea99e9e4740012f8ebc4d02f2c/example/ios/Runner.xcodeproj/project.pbxproj#L361

Let me know if this solves the issue.

ride4sun, 2022-01-11:
The main project references 12.3
Here is the find output:
[image]

remonh87, 2022-01-11:
No that should be fine. Maybe uncomment 

`# platform :ios, '9.0'` in your pod file and set it to 11.0 (or 12.3 )?

ride4sun, 2022-01-11:
Updated everything to 11.0 just to make sure . Did run a clean which is:
```
flutterpod repo update
rm ios/Podfile.lock
rm -rf ios/Pods/
rm .flutter-plugins
rm .packages
rm pubspec.lock
flutter packages upgrade
flutter clean
```

and get the same outcome...

ride4sun, 2022-01-12:
I found on the end of my posfile this:
```
post_install do |installer|
    installer.pods_project.targets.each do |target|
        flutter_additional_ios_build_settings(target)
        target.build_configurations.each do |config|
            config.build_settings['IPHONEOS_DEPLOYMENT_TARGET'] = '9.0'
        end
    end
end
```

I changed the 9 to 11 and that compiles.

Now when the program want to start I get an exception at startup
```
Xcode build done.                                           543.0s
(lldb) dyld[6556]: Symbol not found: _OBJC_CLASS_$_Reachability
  Referenced from: /private/var/containers/Bundle/Application/9524EED3-E189-4E90-904B-13CBBC57D86B/Runner.app/Frameworks/connectivity.framework/connectivity
  Expected in: /private/var/containers/Bundle/Application/9524EED3-E189-4E90-904B-13CBBC57D86B/Runner.app/Frameworks/Reachability.framework/Reachability
* thread #1, stop reason = signal SIGABRT
    frame #0: 0x000000010456a9b4 dyld`__abort_with_payload + 8
dyld`__abort_with_payload:
->  0x10456a9b4 <+8>:  b.lo   0x10456a9d0               ; <+36>
    0x10456a9b8 <+12>: stp    x29, x30, [sp, #-0x10]!
    0x10456a9bc <+16>: mov    x29, sp
    0x10456a9c0 <+20>: bl     0x104531568               ; cerror_nocancel
Target 0: (Runner) stopped.
Installing and launching...                                        36.0s
Error launching application on iPhone 6s.
```

this seemed to be related to this:
https://github.com/flutter/flutter/issues/69159

When I look into my Dart Packages I see both connectivity and connectivity plus.
[image]

Well in short - I am stuck again

ride4sun, 2022-01-12:
After removing the connectivity package all works - this can be closed. Thank you

closed 2022-01-12 by ride4sun