PhilipsHue/flutter_reactive_ble #487: Connect devices that support ANCS

gaogaogoo, 2022-01-11:
**Is your feature request related to a problem? (please describe)**
I use this project to connect a device that supports ANCS service, IOS will require the device to be paired with the system.
1. The paired device will remain connected to the system even if the APP is closed. When the app is reopened, the device cannot be searched.
2. Some users will first connect and pair with the device in the Bluetooth list of system settings. The device cannot be searched when the app is opened.

**Describe the solution you'd like**
I hope to provide devices connected by the system. Provide CBCentralManager.retrieveConnectedPeripheralsWithServices in iOS Bluetooth Library in dart.

Thank you!

Taym95, 2022-06-13:
This goes a bit against our philosophy of not introducing platform specific methods, looks like IOS specific method.

closed 2022-06-13 by Taym95

liang-notes, 2022-08-04:
@gaogaogoo Hi, How do you solved this issue?