PhilipsHue/flutter_reactive_ble #497: Unsupported status on iOS Simulator

pieterbergmans, 2022-01-19:
Hi...I'm getting an ```unsupported``` BleStatus with the following code on iOS simulator.  Is this supposed to work on iOS simulator?  Sorry, I feel like this is a dumb question because iOS simulator doesn't support Bluetooth but I'm not finding any info in the docs regarding how else to test this library.
```
Future<void> isBleReady() async {
    _ble.statusStream.listen((event) async {
      await Future.delayed(const Duration(milliseconds: 3500));
      log.i('ble listening: ${event.toString()}');
      if (event == BleStatus.ready) {
        log.i('ble is ready');
      }
    });    
  }

@override
  Widget build(BuildContext context) {
    return Container(
      child: BusyButton(
        title: 'Scan',
        onPressed: () {
          isBleReady();
        },
      ),
    );
  }
```

sonelektrikci, 2022-01-20:
You can not use Bluetooth on ios simulators. You have to use real devices. So it's not package error.

pieterbergmans, 2022-01-20:
@sonelektrikci - thanks for confirming.  I plugged in my phone and all works good now!

closed 2022-01-31 by Taym95