PhilipsHue/flutter_reactive_ble #498: AndroidManifest.xml clashes with other packages

dusalex, 2022-01-19:
**Describe the bug**
When trying to upload to the Google Play store, Apps build with flutter_reactive_ble may be rejected because of multiple definitions of e.g. "android.permission.ACCESS_COARSE_LOCATION" or "android.permission.ACCESS_FINE_LOCATION" with different maxSdkVersion.

**To Reproduce**
Steps to reproduce the behavior:
Build an App with the plugins flutter_reactive_ble and e.g. [location](https://pub.dev/packages/location) and try to upload the aab file to the Google Playstore. The file will be rejected, because the abovementioned permissions are defined twice in the merged AndroidManifest.xml, once with and once without maxSdkVersion. 

**Expected behavior**
Uploading apps build with flutter_reactive_ble and other packages to the Play Store should work.

**Additional context**
To avoid such clashes, the packages should not include the permission settings in their AndroidManifest.xml files but instead instruct the user to set these values in their app-level manifests.

dusalex, 2022-01-25:
Here's the description of a workaround to remove the clashing permissions: https://github.com/flutternetwork/WiFiFlutter/issues/237

I still suggest to either remove the permissions in the package's AndroidManifest.xml, or, alternatively, describe this workaround in the package documentation.

Taym95, 2022-01-31:
I think it is already documented in README https://github.com/PhilipsHue/flutter_reactive_ble#which-permissions-are-needed

closed 2022-02-02 by PieterAelse

dusalex, 2022-02-02:
I don't think it's properly covered in the readme file yet. The readme file only explains how to remove the BLUETOOTH_SCAN permission. The problem I was referring to is caused by the following two entries in the package's AndroidManifest.xml:

```
<!-- required for API 23 - 30 -->
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        android:maxSdkVersion="30" />
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_FINE_LOCATION"
        android:maxSdkVersion="30" />
```

This clashes in the merged manifest if other packages are included which also use these permissions (but without the maxSdkVersion), e.g., the wifi_iot package.

In such a case, these permissions need to be removed from the merged manifest by adding the following lines to the AndroidManifest.xml of the own app.

```
<!-- remove the permission mentioned in ble lib -->
<uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        android:maxSdkVersion="30"
        tools:node="remove" />
<uses-permission-sdk-23
        android:name="android.permission.ACCESS_FINE_LOCATION"
        android:maxSdkVersion="30"
        tools:node="remove" />
```

In addition to the lines to remove the permissions, it is important to also add the "tools" to the manifest tag at the very top of the AndroidManifest.xml file:

```<manifest xmlns:android="http://schemas.android.com/apk/res/android" package="com.example.exampleapp" xmlns:tools="http://schemas.android.com/tools">```

Rexios80, 2022-02-10:
Yeah the right solution for this it to not include those permissions and have the user include them in their own manifest. This issue is quite hard to track down unless you know what you're doing.

For example: https://pub.dev/packages/polar