PhilipsHue/flutter_reactive_ble #499: Cash on connect on IOS but just fine on Android

ride4sun, 2022-01-19:
**Describe the bug**
The app is crashing on IOS when I call connect. I used pretty much the code from the example which works fine on Android with no Issues but crashes on IOS

This is my connect method which called the connector like in the example:
```
Future connect(GenericDevice device) async {
    loading.value = true;
    if (_connected.value) {
      _log.fine('disconnect - from connect, already connected');
      _disconnect();
    }
    if (isScanning) await _stopScan();

    _discoveredDevice = _devices.value.firstWhere(
        (bleDevice) => device == bleDevice as GenericDevice,
        orElse: () => null);

    //the device might be not present anymore
    if (_discoveredDevice.isNotNull) {
      await _connector.connect(device.id);
    } else {
      _disconnect();
      infoInternal.value = 'Could not connect to ${device?.name}';
    }

    loading.value = false;
  }
```
with the connector being:
```
 Future<void> connect(String deviceId) async {
    _logMessage('Start connecting to $deviceId');
    _connection = _ble.connectToDevice(id: deviceId).listen(
      (update) {
        _logMessage(
            'ConnectionState for device $deviceId : ${update.connectionState}');
        _deviceConnectionController.add(update);
      },
      onError: (Object e) =>
          _logMessage('Connecting to device $deviceId resulted in error $e'),
    );
  }
```

Here is a stacktrace:
```
flutter: model.bloc_base.dart - LOADING -> ble_uart_bluetooth _connect
flutter: stream.stream_value.dart name: ble_uart_bluetooth _connect value: false changed to: true
flutter: btBloc.flutter_ble_lib_bloc.dart Stop ble discovery
flutter: btBloc.flutter_ble_lib_bloc.dart Start connecting to 9DA5C482-6A75-9A21-4E09-29A31E45D19B
Warning! No event channel set up to report a connection update
reactive_ble_mobile/PluginController.swift:90: Assertion failed
reactive_ble_mobile/PluginController.swift:90: Assertion failed
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x00000001893d732c libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x1893d732c <+0>: ret    
libswiftCore.dylib`_swift_reportToDebugger:
    0x1893d7330 <+0>: b      0x1893d732c               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x1893d7334 <+0>: adrp   x8, 439371
    0x1893d7338 <+4>: ldrb   w0, [x8, #0x384]
Target 0: (Runner) stopped.
Lost connection to device.
Failed to send request: {"jsonrpc":"2.0","id":"911","method":"resume","params":{"isolateId":"isolates/68101936672823"}}
```

A clear and concise description of what the bug is.

**To Reproduce**
**Smartphone / tablet**
 - Device: [e.g. iPhone 6s]
 - OS: IOS 15.1

**Peripheral device**
- OBDII Carista Bluetooth Adapter

**Configuration
```
jay@mini obdchk % flutter doctor -v
[✓] Flutter (Channel stable, 2.8.1, on macOS 11.6 20G165 darwin-x64, locale en-US)
    • Flutter version 2.8.1 at /Users/jay/development/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 77d935af4d (5 weeks ago), 2021-12-16 08:37:33 -0800
    • Engine revision 890a5fca2e
    • Dart version 2.15.1

[✓] Android toolchain - develop for Android devices (Android SDK version 29.0.2)
    • Android SDK at /Users/jay/Library/Android/sdk
    • Platform android-29, build-tools 29.0.2
    • Java binary at: /Applications/Android Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7281165)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 13.2.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • CocoaPods version 1.11.2

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2020.3)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.10+0-b96-7281165)

[✓] VS Code (version 1.60.2)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.12.2

[✓] VS Code (version 1.44.2)
    • VS Code at /Users/jay/Downloads/Visual Studio Code-2.app/Contents
    • Flutter extension version 3.12.2

[✓] VS Code (version 1.42.1)
    • VS Code at /Users/jay/Downloads/Visual Studio Code 2.app/Contents
    • Flutter extension version 3.12.2

[✓] VS Code (version 1.41.1)
    • VS Code at /Users/jay/Downloads/Visual Studio Code.app/Contents
    • Flutter extension version 3.12.2

[✓] Connected device (2 available)
    • iPhone 6s (mobile) • 10319c5335ce8ef1b6ee16da68ffcc00e4c56493 • ios            • iOS 15.0.1 19A348
    • Chrome (web)       • chrome   
```

remonh87, 2022-01-20:
did you add `    WidgetsFlutterBinding.ensureInitialized();` to your main.dart?

ride4sun, 2022-01-26:
I did that in the main package but it seems like I have to do that as well in the Bluetooth package. That fixed the problem. Thx

closed 2022-01-26 by ride4sun

vaibhiarora03, 2022-03-17:
Hi @ride4sun. what exact change did helped you out to resolve this issue? In which file of the package you added `WidgetsFlutterBinding.ensureInitialized();`?

tang158152, 2024-07-29:
[image]
@remonh87  
[image]  { assert(false); return }