PhilipsHue/flutter_reactive_ble #500: Reconnection of device and Foreground Service

myogesh7, 2022-01-20:
Can I run this application continuously in foreground for whole night or is there any other way to achieve it.
 
And Does this library auto connect the device and restart data flow.

Taym95, 2022-01-31:
I don’t understand what do you mean, library does not auto-connect, you need to implement this on your app

rohitsangwan01, 2022-02-23:
@myogesh7 you can use this plugin for foreground Tasks flutter_foreground_task[https://pub.dev/packages/flutter_foreground_task] 
For android you can use Ignoe_Battery_optimisation option , and App will keep running in Background until you swipe up the app , and if you want the ble Connection to disconnect with device on swipUp app from memory , you have to add few Extra steps  , and its working for IOS as well , just Add UiBackgroundModes for bluetooth and all , and app will keep working in background if you are continously doing something with bluetooth , but ios pauses app automatically if your app is not performing any UIBackgroundMode(xhttps://developer.apple.com/documentation/bundleresources/information_property_list/uibackgroundmodes) ,well there are few hacks for this as welll
and library will not autoConnect once you terminate stream , or disconnect from device somehow and reopen app , you might have to add your own logic for that

Taym95, 2022-06-13:
I will close this issue for now, since no active anymore

closed 2022-06-13 by Taym95