PhilipsHue/flutter_reactive_ble #502: Check if Bluetooth open or not.

sonelektrikci, 2022-01-21:
**Is your feature request related to a problem? (please describe)**
When I wanted to add some functionality to my UI, it's almost impossible to check if bluetooth service is enabled or disabled. Handling the problem with BleStatus is nearly impossible for every situtation.

**Describe the solution you'd like**
A function can be coded to only check if bluetooth enabled or disbled.

**Describe alternatives you've considered**
I tried the handle the problem with BleStatus, it's nearly impossible to handle.
Also saw flutter_blue has a feature like this.  (flutterBlue.isOn)

oooldCaption, 2022-01-25:
[image]
 it's work; 
What is your physical device system version? iOS or Andorid?

sonelektrikci, 2022-01-25:
> [image] it's work; What is your physical device system version? iOS or Andorid?

It works but it does not stay on this status everytime. For example, if the location permission did not granted; status == BleStatus.unauthorized. So you can't check if bluetooth is enabled or not. I want to check the bluetooth enabled-disabled constantly. I found this solution tho;

https://github.com/PhilipsHue/flutter_reactive_ble/issues/426#issuecomment-953631490

I changed the code a little bit and only checked if adapter.isEnabled.

Taym95, 2022-01-31:
As @remonh87 said:

We as maintainers had a discussion and decided to not put a method in this library. For the following reasons:

We initialise things lazily so requesting a system dialog on iOS on initialise is not a good option
There are a lot of other packages that solve this problem. For example the package mentioned by @rohitsangwan01
and I personally like the pub.dev/packages/app_settings .

closed 2022-01-31 by Taym95