PhilipsHue/flutter_reactive_ble #507: Android pairing request popup appears twice

BradKwon, 2022-01-27:
**Describe the bug**
When a bluetooth peripheral, which has a Nordic chip (I am not sure this happens only with this chip though), is connected to some Android phones, the Android native pairing request popup appears twice. The first popup does nothing.

**To Reproduce**
Steps to reproduce the behavior:
1. Run example Android app
2. Connect the bluetooth peripheral, possibly with a Nordic chip.
3. Check if Android native pairing request popup appears.
4. Click "pair" button on it.
5. Observe the 2nd pairing popup appears.

**Expected behavior**
The Android native pairing popup appears only once.

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Google Pixel 4a
 - OS: Android 12
 - Package version: 5.0.1

**Peripheral device**
- Vendor, model: Custom with Nordic chip
- Does it run a custom firmware / software: no

**Additional context**
This also happens with the general BLE scanner app however, I am wondering if there are any workaround for it.

PieterAelse, 2022-02-02:
Can you reproduce the issue with any other BLE device than the custom Nordic chip? As we're not seeing this issue in the field, it might have to do with your device (even though the nRF Connect app doesn't show it).

And to be precies; what "general BLE scanner app" do you mean?

BradKwon, 2022-02-03:
@PieterAelse I have tested it again with other BLE mouse. I think this seems not a Nordic chip. I was not able to find which chip this has. The pairing request popup appears twice as well with it. 

I noticed that it appears only once on my two Android 9 devices, but on the Android 12 device, Google Pixel 4a.

**_My Android 9 devices:_**
Motorola X4
LG G2 w/ Lineage OS

I mean the nRF Connect app by "general BLE scanner app". Sorry for the confusion. I just used the term in the template.

Based on my further test, the issue only happens on Android 12. Unfortunately, I have only Google Pixel 4a so cannot test it on other Android 12 devices. And the nRF Connect app also shows it twice so it looks like a general issue.

It would be nice someone else report this or can test it on other Android 12 devices. And it would be nice if there are any workaround.

GaelleJoubert, 2022-02-09:
I have the same thing happening to me. 
On a Pixel 4A5G with Android 12. 
If I don't accept it twice, it's not working afterwards. 
I try my code on two other phone :
- Pixel 4A with Android 11
- RugGear with *Android 8 
And I don't have this issue of "double pop up".
So it looks like it is related to android 12.

gourav6m17, 2022-04-25:
@jhewitt Have your find any solution actually, I'm also facing same issue. If found, it's helpful for everyone.

martijnvanwelie, 2022-04-25:
The double pairing popup is an Android bug. It happens when the remote peripheral initiates the bonding process. The workaround is to initiate bonding yourself. This can be done on Android using createBond() but it cannot be done iOS.

BradKwon, 2022-04-30:
@martijnvanwelie Thanks for the comment. I thought it would be good to get this around via this package however, it seems that there is no workaround as I haven't got any feedback here since Feb. 2. 

Then is it possible for you to provide me with how I can initiate the Android bonding with this package in Flutter?

dusalex, 2022-06-15:
The same issue also happens when connecting to an ESP32 where the characteristics access requires pairing/bonding. Based on the discussion above, it sounds like adding the option to manually call the createBond() method on Android may resolve the issue?

BradKwon, 2022-08-24:
Hi @PieterAelse, I already answered back in Feb. and the label says still "awaiting response". Do you have any ideas what is the problem here?

Taym95, 2022-10-24:
> The same issue also happens when connecting to an ESP32 where the characteristics access requires pairing/bonding. Based on the discussion above, it sounds like adding the option to manually call the createBond() method on Android may resolve the issue?

We would rather not, because we don't want to have platform code specific inside this library.

dusalex, 2022-10-24:
@Taym95 Hmm... there are already Android-specific operations, as listed in the [docs](https://pub.dev/documentation/flutter_reactive_ble/latest/), e.g., requestMtu, requestConnectionPriority, clearGattCache. As there are so many difference regarding BLE between Android and iOS, a library addressing this topic somehow needs to be able to properly manage them in my opinion.

paintingStyle, 2022-12-09:
> 
So how should this problem be solved?
How to manage them properly, you didn't make it clear, I don't know how to do it?
Looking forward to your reply

pzehle, 2022-12-21:
I am having the same issue. Any way to solve it?

ben91187, 2023-03-15:
Ist there any update to this issue? I am facing the same issue.

exilat, 2023-04-21:
This issue was reproduced with several embedded devices: espressif ESP32 and Thunderboard Sense2. Also with a host of other devices Motorola G30, Huawei P30 Light, all with Android Version 12. 
We could find a workaround on the embedded device side for this. The issues , or the second pop-up is appearing when you enable encryption on the GATT server characteristics. If you disable encryption requirement, the second pop-up disappears.
So android when discovers charactheristichs that need encryption pushes another pop-up. Which is not displayed all the time. 
Also the device appears in the bonded list even though the device is set up to only pair.
This workaround is not a proper fix. We are reducing security requirement only to avoid this problem. However people facing this issue could in theory avoid it like this.

clemens-, 2023-06-13:
I would also like to have a solution here. And there is already platform-specific code in this library anyway. Ideally, you could argue that you don't want platform-specific **workflows** instead of no platform-specific **code**.

SatyendraM1990, 2023-08-30:
@exilat  I am facing the same problem with android 13 and android 12 device.
Please suggest any solution to fix this issue.

clemens-, 2023-08-31:
I circumvented this problem by connecting to the device and then waiting until the device shows up in the paired devices list. This means only one pairing request will be sent, after that the app is sent into a loop until the user has paired the device. It's crude, but it works. I will still need to add a loop escape sequence. I used a [Method Channel](https://docs.flutter.dev/platform-integration/platform-channels) to do that. Here is how it looks like in code:

Flutter code for connecting and waiting on pairing state:
```java
                // Trigger bonding on Android
                if (Platform.isAndroid) {
                  bool paired =
                      await Provider.of<BleService>(context, listen: false)
                          .checkPairingState();
                  print("Pairing state: $paired");
                  if (paired == false) {
                    await Provider.of<BleService>(context, listen: false)
                        .connectMyDevice();
                    do {
                      paired =
                          await Provider.of<BleService>(context, listen: false)
                              .checkPairingState();
                    } while (paired == false);
                    await Provider.of<BleService>(context, listen: false)
                        .disconnect();
                  }
                }
```

checkPairingState method
```java
  static const platform = MethodChannel('ch.myapp.app/bluetooth');

  Future<bool> checkPairingState() async {
    return true;
    try {
      var pairedDevices = await platform.invokeMethod('getPairedDevices');
      print("pairedDevices: $pairedDevices");
      for(String pairedDevice in pairedDevices) {
        if(pairedDevice == _prefs.getString('deviceId')) {
          return true;
        }
      }
    } on PlatformException catch (e) {
      print('PlatformException: ${e.code} ${e.message}');
    }
    return false;
  }
```

MainActivity.kt
```java
    override fun configureFlutterEngine(flutterEngine: FlutterEngine) {
        super.configureFlutterEngine(flutterEngine)
        MethodChannel(flutterEngine.dartExecutor.binaryMessenger, "ch.myapp.app/bluetooth").setMethodCallHandler {
                call, result ->
            if(call.method == "getPairedDevices") {
                val bluetoothManager: BluetoothManager = getSystemService(BluetoothManager::class.java)
                val bluetoothAdapter: BluetoothAdapter? = bluetoothManager.adapter
                if (bluetoothAdapter == null) {
                    result.notImplemented()
                }
                if (ActivityCompat.checkSelfPermission(
                        this,
                        Manifest.permission.BLUETOOTH_CONNECT
                    ) != PackageManager.PERMISSION_GRANTED
                ) {
                    result.error("Permission missing", "BLUETOOTH_CONNECT not granted", null)
                } else {
                    val pairedDevices: Set<BluetoothDevice>? = bluetoothAdapter?.bondedDevices
                    val pairedDevicesList: MutableList<String> = ArrayList()
                    pairedDevices?.forEach { device ->
                        pairedDevicesList.add(device.address.toString())
                    }
                    result.success(pairedDevicesList)
                }
            }
            else {
                result.notImplemented()
            }
        }
    }
```

mrpurpleknight, 2023-10-18:
@clemens- can you give a more details regarding the dart calls? For example, can you share the connectMyDevice() method of your BleService? I'm trying to implement a similar logic but I still got 2 popups

dianke0201, 2024-04-11:
This seems need to be done on BLE peripheral device side, since almost every BLE communication will enable charateristic encryption, it will force the central device send out Pairing Request, while if BLE peripheral device send Security Request after connection up, it will trigger another Pairing Request, testing with TI CC2642R2 device, if I **remove the Security Request**, then only one Pairing pops out.

vontdeux, 2024-09-25:
> @mrpurpleknight the trick is to start connecting so that a pairing request is triggered, but then periodically check if it has been accepted. I added some more comments here:
> 
> ```java
>                 // Trigger bonding on Android
>                 if (Platform.isAndroid) {
>                 // Check the initial pairing state
>                   bool paired =
>                       await Provider.of<BleService>(context, listen: false)
>                           .checkPairingState();
>                   print("Pairing state: $paired");
>                   // If we are already paired, we can skip this next section
>                   if (paired == false) {
>                     // If we are not yet paired, start connecting to trigger a pairing request
>                     await Provider.of<BleService>(context, listen: false)
>                         .connectMyDevice();
>                     // Now enter a loop until the user has paired the device. 
>                     // This prevents more pairing requests in the meantime
>                     do {
>                       paired =
>                           await Provider.of<BleService>(context, listen: false)
>                               .checkPairingState();
>                     } while (paired == false);
>                     // Disconnect, as I only want to pair at this stage.
>                     // The next Widget can always start with a disconnected and paired device
>                     await Provider.of<BleService>(context, listen: false)
>                         .disconnect();
>                   }
>                 }
> ```
> 
> Note that for good practise you should add a timer or something that prevents the loop from running for all eternity. That I have not done yet here.
> 
> Also, here is the rest of the code you requested, but I don't think it will help you, as all the magic is in the code already posted:
> 
> ```java
> // Connect to the stored device
>   Future<void> connectMyDevice() async {
>     String? deviceId = _prefs.getString('deviceId');
>     if (deviceId == null) {
>       return;
>     }
>     connect(deviceId);
>   }
> 
>   // Connect to a device
>   Future<void> connect(String deviceId) async {
>     print('Start connecting to $deviceId');
>     _prefs.setString('deviceId', deviceId);
>     await cancelNotifications();
>     _connection = _flutterReactiveBle.connectToDevice(id: _prefs.getString('deviceId')!).listen(
>       (update) async {
>         _connectionStateUpdateHandler(update.connectionState);
> 
>         if (update.connectionState == DeviceConnectionState.connected) {
>           _setCharacteristics();
>         }
>       },
>       onError: (Object e) {
>         print('Connecting to device ${_prefs.getString('deviceId')} resulted in error $e');
>       },
>     );
>   }
> ```

is "BleService" is a class that you wrote or it is part of flutter_reactive_ble?

rainerfritz, 2025-04-03:
It still behaves the same in the actual release of the library. In my case the ESP32 acts as BLE server and the Android phone (v14) as peripheral.