PhilipsHue/flutter_reactive_ble #509: writeCharacteristic does not respect service UUID in Android implementation

dusalex, 2022-01-29:
**Describe the bug**
When interacting with a BLE device that has the same characteristic UUIDs under different service UUIDs, the write characteristic commands do not ensure writing to the correct characteristic, even though the supplied parameter of type QualifiedCharacteristic contains the correct combination of service UUID and characteristic UUID.

**Expected behavior**
Addressing a characteristic with a unique combination of characteristic UUID and service UUID should be possible.

**Additional context**
https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/packages/reactive_ble_mobile/android/src/main/kotlin/com/signify/hue/flutterreactiveble/ble/ReactiveBleClient.kt

For example, lines 176 - 186:
```
override fun writeCharacteristicWithResponse(
        deviceId: String,
        characteristic: UUID,
        value: ByteArray
    ): Single<CharOperationResult> =
        executeWriteOperation(
            deviceId,
            characteristic,
            value,
            RxBleConnection::writeCharWithResponse
        )
```

As the function only takes the charactistic UUID but not the service UUID, it *cannot* ensure writing to the correct characteristic in cases where the same characteristic UUID is present in multiple services. (The same holds for other functions in this file.)

On the other hand, the iOS implementation makes sure to respect the service:
https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/packages/reactive_ble_mobile/ios/Classes/ReactiveBle/Tasks/CharacteristicWrite/CharacteristicWriteTaskController.swift

```
guard
    let service = peripheral.services?.first(where: { $0.uuid == task.key.serviceID }),
    let characteristic = service.characteristics?.first(where: { $0.uuid == task.key.id }),
    characteristic.properties.contains(.write)
```

ertan2002, 2022-01-31:
same problem here

OlivierToussaint, 2022-03-11:
Same probleme here

spkersten, 2022-11-08:
Even within the same service, there may be multiple characteristic definitions with the same characteristic uuid. Spec:
> A service may have multiple characteristic definitions with the same Characteristic UUID.

This is related to #478

PR #776 (2023-08-28): Support multiple services or characteristics with the same id
Fixes #478
Fixes #509

The change is mostly backwards compatible. The exception to that is that if you try to use the existing read/write/subscribe methods for a characteristic ids for which there is no single unique instance, an exception is thrown. Before, it would read/write/subscribe the first matching characteristic.
closed 2023-08-28 by Taym95