PhilipsHue/flutter_reactive_ble #511: Failed to compile with SDK<31

SteMMo33, 2022-02-02:
Hi all,
I'm developing a sample app to test the plugin.
Since I'm compiling againt SDK 30 I added the patch suggested to avoid a known problem:

    <uses-permission
        android:name="android.permission.BLUETOOTH_SCAN"
        android:usesPermissionFlags="neverForLocation" tools:node="remove" />

Now I have other problem during compilation:

`FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:generateReleaseBuildConfig'.
> Failed to calculate the value of task ':app:generateReleaseBuildConfig' property 'buildConfigPackageName'.
   > Failed to query the value of property 'packageName'.
      > org.xml.sax.SAXParseException; systemId: file:/C:/Progetti/flutterProjects/blue_flutter/android/app/src/main/AndroidManifest.xml; lineNumber: 13; columnNumber: 78; Il prefisso "tools" per l'attributo "tools:node" associato a un tipo di elemento "uses-permission" non � associato.`

The meaning is that (my free translation..): the prefix 'tools' for the associated attribute "tools:node" associated to element "uses-permission" is not associated. 

This is the doctor output:

    [√] Flutter (Channel stable, 2.5.1, on Microsoft Windows [Versione 10.0.19044.1503], locale it-IT)
    [!] Android toolchain - develop for Android devices (Android SDK version 30.0.3)

Any idea?

PieterAelse, 2022-02-02:
Do you have the `tools` namespace declared at the top of your `AndroidManifest.xml` file?

Example:
```xml
<?xml version="1.0" encoding="utf-8"?>
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    xmlns:tools="http://schemas.android.com/tools"
    package="your.package.name">
    ....
</manifest>
```

SteMMo33, 2022-02-02:
Yes, you are right.
Sorry, I missed ... no it is OK!
Thanks!

closed 2022-02-02 by SteMMo33