PhilipsHue/flutter_reactive_ble #512: Different DiscoveredDevice.id on Android and iOS.

unoravi, 2022-02-04:
```
final flutterReactiveBle = FlutterReactiveBle();

flutterReactiveBle.scanForDevices(withServices: [],scanMode: ScanMode.lowLatency).listen((device) async {
  print(device);
});
```

Above code snippet gives different results on Android and iOS devices as shown below.

Android DiscoveredDevice Properties
```
DiscoveredDevice(id: AC:23:3F:A3:69:06, name: , serviceData: {0000ffe1-0000-1000-8000-00805f9b34fb: [161, 3, 100, 0, 5, 255, 254, 0, 245, 6, 105, 163, 63, 35, 172]}, serviceUuids:[0000ffe1-0000-1000-8000-00805f9b34fb], manufacturerData: [], rssi: -32)
```

iOS DiscoveredDevice Properties
```
DiscoveredDevice(id: CF216209-4D89-CA6A-7FEF-5CE4EFA8DFFF, name: , serviceData: \{ffe1: [161, 3, 100, 0, 5, 255, 254, 0, 245, 6, 105, 163, 63, 35, 172]\}, serviceUuids: [ffe1], manufacturerData: [], rssi: -46)
```

Both results has **different DiscoveredDevice.id values**. So how can I found that both results are of same BLE device or not ? Actual DiscoveredDevice.id value should be AC:23:3F:A3:69:06 which is not found on iOS properties.

OS & Dependency Version
- iOS 15 & 16
- Android 11
- flutter_reactive_ble: ^5.0.2

vin-lz, 2022-02-04:
This is something that iOS does. IOS hides the MAC address of you BLE device and provides UUID instead.

unoravi, 2022-02-04:
@vin-lz 
In this case, Android should show UUID of BLE Device. At least there should be one common id to recognize BLE device on Android/iOS at same time.

Taym95, 2022-02-07:
This is not possible, BLE ID in IOS and Android are different, this is how the BLE stack is implemented in IOS, the library can do nothing about it.

closed 2022-02-07 by Taym95

remonh87, 2022-02-23:
@unoravi  this has nothing to do with our library (in fact you will experience the same everywhere). iOS obscures the Mac address for security purposes and generates a randomised uuid out of it.

unoravi, 2022-03-11:
@remonh87 
I have list of device ids and uuids. I want to execute particular action on movement of each device. Each device has different actions so ID wise identification required in this case.

remonh87, 2022-03-11:
So I think you need to specify a unique id on the peripheral itself i you want to share it across both platforms. The device id is not usable for the above reasons

unoravi, 2022-03-11:
@remonh87 
Peripheral itself has a unique ID. Android is able to retrieve same ID but iOS is unable to retrieve that ID.