PhilipsHue/flutter_reactive_ble #514: connectToDevice method  and autoconnect issue

compvisiondeveloper, 2022-02-09:
**Describe the bug**
 I called the 'connectToDevice' and then listen stream . 'listen' method was closed normally when my BLE device was disconnected.But didn't connet to automaticly despite to without 'connectiontimeout' parameter.

**My Script**

Uuid _UART_UUID = Uuid.parse("0000fff0-0000-1000-8000-00805f9b34fb");
Uuid _UART_RX = Uuid.parse("0000fff2-0000-1000-8000-00805f9b34fb");
Uuid _UART_TX = Uuid.parse("0000fff1-0000-1000-8000-00805f9b34fb");
Map<Uuid, List<Uuid>> servicesWithCharacteristicsToDiscover ={_UART_UUID: [_UART_TX, _UART_RX]};

     _currentConnectionStream = flutterReactiveBle.connectToDevice(id: indeX,
      servicesWithCharacteristicsToDiscover: servicesWithCharacteristicsToDiscover,
     );

    Subscription = _currentConnectionStream.listen((event) {

      switch (event.connectionState) {
        case DeviceConnectionState.connecting:
          {
            print("connecting");
            break;
          }
        case DeviceConnectionState.connected:
          {
            index = indeX;
            DeviceName = deviceName;
            deviceID = indeX;
            DeviceNo = deviceNo;
            ConnnecionStateStream = _currentConnectionStream;
            connectionState = true.obs;
            break;
          }
        case DeviceConnectionState.disconnecting:
          {
            print("disconnecting");
            break;
          }
        case DeviceConnectionState.disconnected:
          {
            print("disconnected");
            connectionState = false.obs;
            break;
          }
      }
    }, onDone: () {
          print("ONDONE");
          connectionState = false.obs;
    });

 **OR**
      I tried like this,but happend the same issue.
     _currentConnectionStream = flutterReactiveBle.connectToDevice(
                          id: indeX,
      );

**Smartphone / tablet**
 - Device1:  Huawei AGR-W09 
  - Device:2  SAMSUNG GALAXY TAB A7 SM-T500  issue valid for two device.

remonh87, 2022-02-09:
@FurkanTskn  the android `autoconnect` flag is very confusing I think Google made a mistake naming it like that. What the autoconnect is doing is it will connect to the Ble device when it is available to connect to. In other words it never times out. 

When you you do not use the flag it will directly try to connect and depending on the Ble stack inmplementation of the phone time out.   See https://developer.android.com/reference/android/bluetooth/BluetoothDevice#connectGatt(android.content.Context,%20boolean,%20android.bluetooth.BluetoothGattCallback). 

In general `autoconnect` is slow but stable and without autoconnect is fast but can be flaky. It really depends on your phone and peripheral implementation which one is better. 

Hope that this makes it clear.

closed 2022-02-09 by remonh87

FurkanTskn, 2022-02-09:
thank you for explain @remonh87

FurkanTskn, 2022-02-09:
@remonh87  
Thank you for explain.