PhilipsHue/flutter_reactive_ble #516: Error Disconnecting To Device in iOS

yuriboeira11tx, 2022-02-10:
**Describe the bug**
The documentation explains that to disconnect from a device, just cancel the subscription with deviceSubscription, for example. However, on iOS, the following logs are being triggered:

```
Error unsubscribing from notifications: PlatformException(reactive_ble_mobile.Central.(unknown context at $104e330e4).Failure:1, The operation could not be completed.(reactive_ble_mobile.Central.(unknown context at $104e330e4).Failure erro 1.), {}, null)
```

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one using `connectToDevice();`
2. Disconnecting to device with `await deviceSubscription!.cancel();`
3. Observe a failure with exception: ```
Error unsubscribing from notifications: PlatformException(reactive_ble_mobile.Central.(unknown context at $104e330e4).Failure:1, The operation could not be completed.(reactive_ble_mobile.Central.(unknown context at $104e330e4).Failure erro 1.), {}, null)
```

**Smartphone / tablet**
 - Device: iPhone XS
 - OS: iOS 15.2
 - Package version: 5.0.2

**Peripheral device**
- Does it run a custom firmware: yes

remonh87, 2022-02-14:
@yuriboeira11tx  does this happen all the time or occasionally?

tomstokes, 2022-02-15:
I'm experiencing the same issue. Error occurs on every disconnect, only on iOS devices. Using package version 5.0.2.

bretmh, 2022-02-16:
I am experiencing this as well.

tomstokes, 2022-02-16:
Did some quick debugging.

On my iOS device, the `disconnectFromDevice` platform method [(here)](https://github.com/PhilipsHue/flutter_reactive_ble/blob/c25bedf9bc00abdac60927f42a8aafdc5ae5dbf8/packages/reactive_ble_mobile/ios/Classes/Plugin/SwiftReactiveBlePlugin.swift#L99-L101) gets called before the `stopNotification` platform method [(here)](https://github.com/PhilipsHue/flutter_reactive_ble/blob/c25bedf9bc00abdac60927f42a8aafdc5ae5dbf8/packages/reactive_ble_mobile/ios/Classes/Plugin/SwiftReactiveBlePlugin.swift#L126-L128)

The `disconnectFromDevice` method ejects the peripheral from `activePeripherals`, which causes the `stopNotification` call chain to fail when trying to find the peripheral to turn off notifications.

I've run out of time to debug this for now. Hopefully someone else can run with these clues.

ozzy1873, 2022-02-20:
This is a major problem for me too, but only on iOS. I have to remove my app from sale in the Apple store until it is resolved.

ozzy1873, 2022-03-12:
After this exception occurs, you can reconnect the device but discoverServices and subscribeToCharacteristic fail. I found the workaround is to disconnect/reconnect the device after it fails the first time.

shobhit14gupta, 2022-03-15:
I am experiencing the same issue.  Is there any update or  workaround to fix this issue?

yuriboeira11tx, 2022-03-16:
@shobhit14gupta not yet.

vaibhiarora03, 2022-03-17:
I am also experiencing the same issue. Has anyone got fix for this issue?

mrRedSun, 2022-05-27:
Still happens to me on iOS 15 and iPadOS 15

GaelleJoubert, 2022-06-15:
I have the same Issue. 
On IOS only as well.

I found something that might be helpfull in another issue : #414 
https://github.com/PhilipsHue/flutter_reactive_ble/issues/414#issuecomment-1013830555

The comment says : "_I had the same error. I was able to fix it by canceling the BLE Characteristic Streams first, adding a delay of a few hundred milliseconds then canceling the ConnectionStateUpdate stream.

example

await characteristicStream?.cancel();
await Future.delayed(const Duration(milliseconds: 300));
await deviceConnectionStateStream?.cancel()_; "

I am in the same situation of having subscribed to a characteristic, I tried the fix, but so far it didn't work ...
But I though that might help some of you.

BikeBr0, 2022-06-21:
Thanks for the sample @GaelleJoubert. I am using the MCUMGR library and found that adding a delay between closing the MCUMGR client and cancelling the connection solved the problem for me.

pawlowskim, 2023-06-13:
Suggestion from @GaelleJoubert works. I had to add delay after canceling characteristic stream before I disconnected from the device.