PhilipsHue/flutter_reactive_ble #517: Please add the "canSendWriteWithoutResponse" functionality to the ios

victorLyw, 2022-02-14:
**Is your feature request related to a problem? (please describe)**
When sending long data using ios packet, I can't tell whether the ble device can continue to write, my ble device only supports "WriteWithoutResponse".

**Describe the solution you'd like**
"@property(readonly) BOOL canSendWriteWithoutResponse;",This property lets you know, if Buffer is filled up or not and can transmit more without response.

**Describe alternatives you've considered**
I've tried adding a 10ms delay to every packet sent, but that doesn't work for long range bluetooth devices.

remonh87, 2022-02-14:
@victorLyw I faced the same problem in my passed projects  as alternative and maybe a better solution is to send a `writeWithResponse` for each ...x packet that I sent to the device.  I found this method more reliable and it works for both iOS and Android.  Let me know what you think.

victorLyw, 2022-02-14:
@remonh87 This method is not suitable for all devices, my bluetooth device does not support "writeWithResponse", and ios uses "writeWithResponse" to get an error. I think the control should be handed over to the developer, and "canSendWriteWithoutResponse" is used to determine whether it can continue to send .

remonh87, 2022-02-14:
@werediver there is no function like that on Android. On the polidea library there is functionality to execute long writes (I think it uses `reliableWrite` of the BluetoothGatt class). We can maybe implement sth like that to keep it consistent across platforms.  What do you think?  

It makes sense to implement it to help users not to overflow the device.

victorLyw, 2022-02-14:
@remonh87 

In my project, the Android platform long data writing is stable, because there is no difference between "writeWithResponse" and "writeWithoutResponse" in Android, but when using the "requestMtu" api, the mtu request result must  -3.

In the ios platform, "writeWithoutResponse" is unreliable, the ios platform provides "canSendWriteWithoutResponse" to determine whether the remote device can send writes without a response.

I think "clearGattCache" can be used as a platform specific action for Android and "canSendWriteWithoutResponse" can also be used as a platform specific property for ios.

Taym95, 2022-02-15:
@victorLyw we want to keep the API consistent across platforms, we want to avoid implementing platform specific API

closed 2022-06-13 by Taym95