PhilipsHue/flutter_reactive_ble #518: iOS Build failed

ObranS, 2022-02-16:
**Describe the bug**
When I run iOS build with 

dependencies:
    flutter_reactive_ble: ^5.0.2

I got errors:

Could not build the precompiled application for the device.
Swift Compiler Error (Xcode): Cannot force unwrap value of non-optional type 'CBPeripheral'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:79:43

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:229:53

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:254:53

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:272:51

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBPeripheral'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Central.swift:272:63

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/ReactiveBle/Tasks/CharacteristicNotify/CharacteristicNotifyTaskController.swift:13:53

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/Plugin/PluginController.swift:292:49

Swift Compiler Error (Xcode): Cannot force unwrap value of non-optional type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/Plugin/PluginController.swift:293:85

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData%20extras/QualifiedCharacteristic.swift:15:45

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBService'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData%20extras/QualifiedCharacteristic.swift:16:48

Swift Compiler Error (Xcode): Cannot use optional chaining on non-optional value of type 'CBPeripheral'
/Users/newfire/.pub-cache/hosted/pub.dartlang.org/reactive_ble_mobile-5.0.2/ios/Classes/BleData%20extras/QualifiedCharacteristic.swift:16:60

**To Reproduce**
Steps to reproduce the behavior:
1. add flutter_reactive_ble: ^5.0.2
2. run the Flutter iOS build

**Expected behavior**
Build successfuly finished

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: iPhone 11
 - OS: iOS 15.2
 - Package version: 5.0.2

ObranS, 2022-02-16:
Besides, I tried to comment package dependencies on the example app and have the same errors.

name: flutter_reactive_ble_example
description: Demonstrates how to use the flutter_reactive_ble plugin.
version: 5.0.2
publish_to: 'none'

environment:
  sdk: '>=2.12.0 <3.0.0'
  flutter: ">=1.10.0"

dependencies:
  flutter:
    sdk: flutter
  flutter_lints: ^1.0.4
  flutter_reactive_ble: ^5.0.2
  functional_data: ^1.0.0
  intl: ^0.17.0

  provider: ^6.0.1

dev_dependencies:
  build_runner: ^2.1.4
  dependency_validator: ^3.1.0
  flutter_test:
    sdk: flutter
  functional_data_generator: ^1.1.2

dependency_overrides:
#  flutter_reactive_ble:
#    path: ../packages/flutter_reactive_ble
#  reactive_ble_mobile:
#    path: ../packages/reactive_ble_mobile

flutter:
  uses-material-design: true

remonh87, 2022-02-16:
it looks like you have an outdated swift version or Xcode version. Can you share the pod file with us? I tried to do the same like you described in the example app and had no issues building it.

ObranS, 2022-02-16:
My Xcode version: Version 13.2.1 (13C100)

Flutter version: 
Flutter 2.10.1 • channel stable • https://github.com/flutter/flutter.git
Framework • revision db747aa133 (7 days ago) • 2022-02-09 13:57:35 -0600
Engine • revision ab46186b24
Tools • Dart 2.16.1 • DevTools 2.9.2

Podfile code is the same as on the example app:

[image]

I've reproduced this issue in the example app (master) just by comment those lines:

[image]

Podfile:

`# Uncomment this line to define a global platform for your project
# platform :ios, '11.0'

# CocoaPods analytics sends network stats synchronously affecting flutter build latency.
ENV['COCOAPODS_DISABLE_STATS'] = 'true'

project 'Runner', {
  'Debug' => :debug,
  'Profile' => :release,
  'Release' => :release,
}

def flutter_root
  generated_xcode_build_settings_path = File.expand_path(File.join('..', 'Flutter', 'Generated.xcconfig'), __FILE__)
  unless File.exist?(generated_xcode_build_settings_path)
    raise "#{generated_xcode_build_settings_path} must exist. If you're running pod install manually, make sure flutter pub get is executed first"
  end

  File.foreach(generated_xcode_build_settings_path) do |line|
    matches = line.match(/FLUTTER_ROOT\=(.*)/)
    return matches[1].strip if matches
  end
  raise "FLUTTER_ROOT not found in #{generated_xcode_build_settings_path}. Try deleting Generated.xcconfig, then run flutter pub get"
end

require File.expand_path(File.join('packages', 'flutter_tools', 'bin', 'podhelper'), flutter_root)

flutter_ios_podfile_setup

target 'Runner' do
  use_frameworks!
  use_modular_headers!

  flutter_install_all_ios_pods File.dirname(File.realpath(__FILE__))
end

post_install do |installer|
  installer.pods_project.targets.each do |target|
    flutter_additional_ios_build_settings(target)
  end
end
`

remonh87, 2022-02-16:
I did that exactly in the example app and had no issues. 

can you run `flutter clean` and then `flutter pub get ` again and build it?

ObranS, 2022-02-16:
> I did that exactly in the example app and had no issues.
> 
> can you run `flutter clean` and then `flutter pub get ` again and build it?

Yes, I did. Besides:

.packeges clean
.pub-spec remove 

for me only working solution:

- copy-paste **reactive_ble_mobile**
- add **override_dependency**:
```
dependency_overrides:
  reactive_ble_mobile:
    path: ./packages/reactive_ble_mobile
```

Possibly it is a Flutter version issue.

Could you share with me recommended (your) Flutter version and channel?
I will update mine with the same and try?

remonh87, 2022-02-16:
flutter doctor
```
Doctor summary (to see all details, run flutter doctor -v):
[✓] Flutter (Channel stable, 2.10.1, on macOS 12.2.1 21D62 darwin-arm, locale en-SE)
[✓] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
[✓] Xcode - develop for iOS and macOS (Xcode 13.2.1)
[✓] Chrome - develop for the web
[✓] Android Studio (version 2021.1)
[✓] IntelliJ IDEA Community Edition (version 2021.2.3)
[✓] VS Code (version 1.64.2)
[✓] Connected device (3 available)
[✓] HTTP Host Availability
```

ObranS, 2022-02-16:
> flutter doctor
>
> ```
> Doctor summary (to see all details, run flutter doctor -v):
> [✓] Flutter (Channel stable, 2.10.1, on macOS 12.2.1 21D62 darwin-arm, locale en-SE)
> [✓] Android toolchain - develop [...]

I found my problem:

[✓] Flutter (Channel stable, 2.10.1, on macOS 11.6.1 20G224 darwin-x64, locale
    en-GB)
[✓] Android toolchain - develop for Android devices (Android SDK version 30.0.2)
[!] Xcode - develop for iOS and macOS (Xcode 12.5.1)
    ! Flutter recommends a minimum Xcode version of 13.
      Download the latest version or update via the Mac App Store.
[✓] Chrome - develop for the web
[✓] Android Studio (version 2020.3)
[✓] VS Code (version 1.62.3)
[✓] Connected device (3 available)
[✓] HTTP Host Availability

Flutter was targeted on wron xcode version. After commands:

`sudo xcode-select --switch /Applications/Xcode.app/Contents/Developer
sudo xcodebuild -runFirstLaunch`

Problem has gone. 

Thank you for your help and sorry that waste your time.

closed 2022-02-16 by Taym95