PhilipsHue/flutter_reactive_ble #529: Issue with subscribeToCharacteristic

abneradekz, 2022-03-04:
When I create a connection using the model below and cancel it, it disconnects but keeps the bluetooth active with the device, is there a way to disconnect completely? 

`
final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: characteristicUuid, deviceId: foundDeviceId);
listen = flutterReactiveBle.subscribeToCharacteristic(characteristic).listen((data) {
      // code to handle incoming data
    }, onError: (dynamic error) {
      // code to handle errors
    });

listen!.cancel()
`

yuriboeira11tx, 2022-03-08:
deviceStream.cancel();

remonh87, 2022-03-09:
which platform? Android or iOS?

abneradekz, 2022-03-10:
Hi

I use Android, haven't started testing on iOS yet.

abneradekz, 2022-03-24:
Hi

I set:
flutterReactiveBle.logLevel = LogLevel.verbose;

Log is:
2022-03-24 15:47:48.929 18342-18505/com.app I/flutter: sub3!.cancel()
2022-03-24 15:47:49.009 18342-18342/com.app D/BluetoothGatt: setCharacteristicNotification() - uuid: 0000fff1-0000-1000-8000-00805f9b34fb enable: false

### Image Connected to device
[image]

### Image after use subscribe.cancel
[image]

### Image after close app
[image]

You see that continue connected after use subscribe.cancel()

cbreezier, 2022-04-23:
@abneradekz In the code you provided, you're canceling the subscription to the characteristic stream.

Where is your code to initially connect to the device? To disconnect from the device, you must cancel the stream from the `connectToDevice` API, not the stream from the `subscribeToCharacteristic` API.

```
final deviceStream = flutterReactiveBle.connectToDevice(id: deviceId);
// After this, the device is connected
deviceStream.listen((status) => {
    print(status)
});

final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: characteristicUuid, deviceId: foundDeviceId);
listen = flutterReactiveBle.subscribeToCharacteristic(characteristic).listen((data) {
// code to handle incoming data
}, onError: (dynamic error) {
// code to handle errors
});
listen!.cancel();

// After this, the device is disconnected
await deviceStream.cancel();
```

abneradekz, 2022-05-03:
```
final deviceStream = flutterReactiveBle.connectToDevice(id: deviceId);
// After this, the device is connected
final subscrib = deviceStream.listen((status) => {
    print(status)
});

final characteristic = QualifiedCharacteristic(serviceId: serviceUuid, characteristicId: characteristicUuid, deviceId: foundDeviceId);
listen = flutterReactiveBle.subscribeToCharacteristic(characteristic).listen((data) {
// code to handle incoming data
}, onError: (dynamic error) {
// code to handle errors
});
listen!.cancel();

// After this, the device is disconnected
await subscrib!.cancel();
```

I changed a little bit in the deviceStatus part, but it really worked that way.

Thanks

Best Regards

closed 2022-05-03 by abneradekz