PhilipsHue/flutter_reactive_ble #530: Scan Fails in Android 12

yuriboeira11tx, 2022-03-08:
**Describe the bug**
Ao entrar no aplicativo e solicitar o scan de dispositivos no Android 12, nenhuma caixa de diálogo de permissão Bluetooth aparece, muito menos de localização. Portanto, venho em meio deste, informar o erro:
```
Device scan fails with error: Exception: GenericFailure<ScanFailure>(code: ScanFailure.unknown, message: "Location Permission missing (code 3)")
```

**Smartphone / tablet**
 - Device: Samsung S10 Plus
 - OS: Android 12
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: Samsung
- Does it run a custom firmware / software: yes

jobfeikens, 2022-03-09:
You need to request the permissions manually: `Permission.bluetoothScan.request()` on Android 12 and `Permission.location.request()` on Android 11 and lower. (Using permission_handler plugin)

flukejones, 2022-03-09:
Currently experiencing the same. I've manually set all the permissions I can think of to try counter this and still no go.

In manifest I set:

```xml
    <!-- Request legacy Bluetooth permissions on older devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH"
        android:maxSdkVersion="30" />
    <uses-permission android:name="android.permission.BLUETOOTH_ADMIN"
        android:maxSdkVersion="30" />

    <!-- Newer devices. -->
    <uses-permission android:name="android.permission.BLUETOOTH_SCAN"/>
    <uses-permission android:name="android.permission.BLUETOOTH_CONNECT"/>
```

and use permission_handler:

```dart
if (Platform.isAndroid) {
      [Permission.bluetooth, Permission.bluetoothScan, Permission.bluetoothConnect].request().then((status) async {
        status.forEach((key, status) async {

          if (status.isDenied ||
              status.isPermanentlyDenied ||
              status.isLimited ||
              status.isRestricted) {
            await showJteConfirmAction(
              context,
              lang.dialogueBLEPermissions.title,
              lang.dialogueBLEPermissions.body + '.\n\n' + lang.openSettings +
                  '.',
            ).then((value) {
              if (value != null && value) {
                openAppSettings();
              }
            });
          }

        });
      });
    }
```

jobfeikens, 2022-03-09:
@flukejones This works for me:
```dart
    if (Platform.isAndroid) {
      final sdk = (await DeviceInfoPlugin().androidInfo).version.sdkInt;

      // Android 12 and higher
      if (sdk >= 31) {
        switch (await Permission.bluetoothScan.request()) {
          case PermissionStatus.granted:
            break;

          case PermissionStatus.permanentlyDenied:
            return completer.complete(UserAction.ENABLE_NEARBY_DEVICES_PERMISSION, () async {
              await AppSettings.openAppSettings();
              return false;
            });

          default:
            return false;
        }

        switch (await Permission.bluetoothConnect.request()) {
          case PermissionStatus.granted:
            break;

          case PermissionStatus.permanentlyDenied:
            return completer.complete(UserAction.ENABLE_NEARBY_DEVICES_PERMISSION, () async {
              await AppSettings.openAppSettings();
              return false;
            });

          default:
            return false;
        }

        // Android 11 and lower
      } else {
        if (await Permission.location.status != PermissionStatus.granted) {

          if (!await completer.complete(UserAction.LOCATION_PERMISSION, _enableLocation)) {
            return false;
          }
        }
      }
    }
```

flukejones, 2022-03-09:
@jobfeikens which phone? Even adjusting to a similar block of code I still get an "Unauthorised"

jobfeikens, 2022-03-09:
@flukejones Google pixel 5. Maybe try reinstalling the app

jhewitt, 2022-03-09:
We are having similar issues with the latest Android 12 builds.  The device we are attempting to bond with requires a PIN code which Android and iOS manage themselves.  We have permissions handled but need a way to determine when bonding has completed.  How do we determine using this PUB when to request services and set up characteristic notifications?  We are encountering the following error when attempting this following the connected event, this same workflow works properly on all iOS versions tested to date.
Note the error response to discovering services:

D/BluetoothAdapter(32345): isLeEnabled(): ON
I/flutter (32345): D	BluetoothServiceImpl._connectDevice:	 connecting 68A
I/flutter (32345): D	SmartLinkRepositoryImpl.connect:	 connect to device
D/BluetoothGatt(32345): connect() - device: 00:A0:50:3E:A6:8A, auto: false
D/BluetoothGatt(32345): registerApp()
D/BluetoothGatt(32345): registerApp() - UUID=affa0af0-9efb-4c48-bec6-c8036b673079
I/flutter (32345): D	SmartLinkRepositoryImpl._connectAndSubscribeToConnectionState.<ac>:	 ble connect state: ConnectionStateUpdate(deviceId: 00:A0:50:3E:A6:8A, connectionState: DeviceConnectionState.connecting, failure: null)
D/BluetoothGatt(32345): onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt(32345): onClientConnectionState() - status=0 clientIf=9 device=00:A0:50:3E:A6:8A
I/flutter (32345): D	SmartLinkRepositoryImpl._connectAndSubscribeToConnectionState.<ac>:	 ble connect state: ConnectionStateUpdate(deviceId: 00:A0:50:3E:A6:8A, connectionState: DeviceConnectionState.connected, failure: null)
I/flutter (32345): D	SmartLinkRepositoryImpl._subscribeToCharacteristics:	 ble Discovering Services
I/flutter (32345): D	SmartLinkRepositoryImpl._subscribeToCharacteristics:	 ble Error: PlatformException(service_discovery_failure, Bonding is in progress wait for bonding to be finished before executing more operations on the device, null, null)
D/BluetoothGatt(32345): onConnectionUpdated() - Device=00:A0:50:3E:A6:8A interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(32345): onConnectionUpdated() - Device=00:A0:50:3E:A6:8A interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(32345): onClientConnectionState() - status=0 clientIf=9 device=00:A0:50:3E:A6:8A
D/BluetoothGatt(32345): close()
D/BluetoothGatt(32345): unregisterApp() - mClientIf=9

closed 2022-03-10 by Taym95

mbabuglia, 2022-03-11:
@Taym95, please reopen this issue. I am having problems with the scan as well. 
My device is a Samsung Galaxy S20 FE with Android 12.

Steps to reproduce:
Run the sample app for the first time. Widget BleStatusScreen shows unauthorized as expected. 
Manually give "Nearby devices" permissions, return to the app.
Status correctly updates, now I can input a service Id and start scanning, but no device is returned, as it should.

The exactly same steps work fine on Android 10 and 11.  
(albeit the permission given manually is Location instead of Nearby devices).

flukejones, 2022-05-31:
In case someone else arrives here: I solved my issue by creating a new flutter project then copying over the main test/libs/ci stuff, and setting up the names etc again. I assume that flutter at some point broke something. Either way it all works now.