PhilipsHue/flutter_reactive_ble #531: Proper workflow for performing device bonding

jhewitt, 2022-03-09:
I originally added this to an existing issue but believe it warrants more visibility.  In addition to the existing Android 12 and security patch problems reported in other issues, we need to determine when a device bonding process has completed based on our observations below.  This is not a problem on all iOS versions tested to date.  We are encountering this problem on a Pixel 5 running Android 12.2 with all security patches applied.

We are having similar issues with the latest Android 12 builds.  The device we are attempting to bond with requires a PIN code which Android and iOS manage themselves.  We have permissions handled but need a way to determine when bonding has completed.  How do we determine using this PUB when to request services and set up characteristic notifications?  We are encountering the following error when attempting this following the connected event, this same workflow works properly on all iOS versions tested to date.
Note the error response to discovering services:

D/BluetoothAdapter(32345): isLeEnabled(): ON
I/flutter (32345): D	BluetoothServiceImpl._connectDevice:	 connecting 68A
I/flutter (32345): D	SmartLinkRepositoryImpl.connect:	 connect to device
D/BluetoothGatt(32345): connect() - device: 00:A0:50:3E:A6:8A, auto: false
D/BluetoothGatt(32345): registerApp()
D/BluetoothGatt(32345): registerApp() - UUID=affa0af0-9efb-4c48-bec6-c8036b673079
I/flutter (32345): D	SmartLinkRepositoryImpl._connectAndSubscribeToConnectionState.<ac>:	 ble connect state: ConnectionStateUpdate(deviceId: 00:A0:50:3E:A6:8A, connectionState: DeviceConnectionState.connecting, failure: null)
D/BluetoothGatt(32345): onClientRegistered() - status=0 clientIf=9
D/BluetoothGatt(32345): onClientConnectionState() - status=0 clientIf=9 device=00:A0:50:3E:A6:8A
I/flutter (32345): D	SmartLinkRepositoryImpl._connectAndSubscribeToConnectionState.<ac>:	 ble connect state: ConnectionStateUpdate(deviceId: 00:A0:50:3E:A6:8A, connectionState: DeviceConnectionState.connected, failure: null)
I/flutter (32345): D	SmartLinkRepositoryImpl._subscribeToCharacteristics:	 ble Discovering Services
I/flutter (32345): D	SmartLinkRepositoryImpl._subscribeToCharacteristics:	 ble Error: PlatformException(service_discovery_failure, Bonding is in progress wait for bonding to be finished before executing more operations on the device, null, null)
D/BluetoothGatt(32345): onConnectionUpdated() - Device=00:A0:50:3E:A6:8A interval=6 latency=0 timeout=500 status=0
D/BluetoothGatt(32345): onConnectionUpdated() - Device=00:A0:50:3E:A6:8A interval=36 latency=0 timeout=500 status=0
D/BluetoothGatt(32345): onClientConnectionState() - status=0 clientIf=9 device=00:A0:50:3E:A6:8A
D/BluetoothGatt(32345): close()
D/BluetoothGatt(32345): unregisterApp() - mClientIf=9

_Originally posted by @jhewitt in https://github.com/PhilipsHue/flutter_reactive_ble/issues/530#issuecomment-1062937242_

jhewitt, 2022-03-11:
I should mention that we have used this PUB successfully on other projects for both iOS and Android with devices that utilize the 'Just Works' pairing workflow.

remonh87, 2022-03-12:
Understandable request and I agree this difficult without a method that make sure bonding is there.  iOS doesn't have a mechanism to determine the bonding state but android does have it. Maybe we should consider exposing an extra check specifically for android where you can request the bonding state of the device. @Taym95  it goes a bit against our philosophy of not introducing platform specific methods. What are your thoughts?

jhewitt, 2022-03-12:
iOS and Android handle the bonding process differently.  Typically bonding takes place when the central device accesses a secure characteristic, at which time the OS takes over and performs BMS handshaking with the peripheral.  Both Android and iOS delay the read or write response until BMS completes.  When a device has all characteristics defined as secure, this occurs during the connection process and iOS delays reporting the connection until BMS is complete where Android does not.

petemillermultimedia, 2023-03-01:
Encountering this exact problem. Older phone and Android version, but the description is the same. Adding this to bump the issue.

invented-pro, 2023-09-09:
Exact same issue on Android.

pzehle, 2023-10-23:
I have this problem on Android as well. I can't find a workaround or anything I can do to get through this error.

chris-otani, 2023-11-21:
I'm running into this issue as well.

enseitankad0, 2024-03-22:
Bump. Pixel 3XL with Android 12 on board, the same issue.

rowdy15, 2024-04-26:
I've got the same issue with android.  I'm trying to use a work around with flutter MethodChannel function calls, but I still haven't got it working exactly how I would like.

enseitankad0, 2024-11-14:
> Same issue with. a customer on OnePlus CPH2409 Android 14

Did you manage to connect using nRF app? If not, the issue is on
manufacturer's side.

RyanTempdrop, 2024-11-14:
@enseitankad0 The issue is on the device of one of Android customer, I have had no issues connecting on our devices, or with nRF.

enseitankad0, 2024-11-14:
You must to try to reproduce on the same device or look for similar issues
with the same CPU (I assume it's Mediatek)

RyanTempdrop, 2024-11-24:
I do not have access to the same device or CPU>. I am not asking for pairing. .And example of a device this is happening on is SM-S921U With Android 14

enseitankad0, 2024-11-24:
I'm not surprised at all. I had the same issues with similar OnePlus Nord
model. I've tried to connect with a Nordic chip using nRF app and it was
impossible. Customer needs to check another phone. This issue is due to
manufacturer's fault. Sth is wrong with Mediatek and BLE driver. This is
not your fault.
If you ask your customer to connect using nRF it will probably fail as well.

jobfeikens, 2025-04-08:
For now, I worked around this problem on my Pixel 5 with the following code: (call using a method channel)
```
@SuppressLint("MissingPermission")
private fun awaitBonding(callId: Int, deviceAddress: String?, result: MethodChannel.Result) {
    if (deviceAddress == null) {
        result.error("ERROR_NO_SUCH_DEVICE", null, null)
        return
    }

    val bluetoothManager = getSystemService(Context.BLUETOOTH_SERVICE) as BluetoothManager
    val device = bluetoothManager.adapter.getRemoteDevice(deviceAddress)

    if (device.bondState != BluetoothDevice.BOND_BONDING) {
        result.success(null)
        return
    }

    val receiver = object : BroadcastReceiver() {
        override fun onReceive(context: Context?, intent: Intent?) {
            val action = intent?.action

            if (action != null && action == BluetoothDevice.ACTION_BOND_STATE_CHANGED) {
                if (!intent.hasExtra(BluetoothDevice.EXTRA_DEVICE)
                        || !intent.hasExtra(BluetoothDevice.EXTRA_BOND_STATE)
                ) {
                    unregisterReceiver(this)
                    result.error("MISSING_EXTRA", null, null)
                    return
                }

                if (!device.equals(intent.getParcelableExtra(BluetoothDevice.EXTRA_DEVICE))) {
                    return
                }

                when (intent.getIntExtra(
                        BluetoothDevice.EXTRA_BOND_STATE, BluetoothDevice.ERROR
                )) {
                    BluetoothDevice.BOND_BONDED, BluetoothDevice.BOND_NONE -> {
                        unregisterReceiver(this)
                        result.success(null)
                    }
                    else -> return
                }
            }
        }
    }
    registerReceiver(receiver, IntentFilter(BluetoothDevice.ACTION_BOND_STATE_CHANGED));
}
```