PhilipsHue/flutter_reactive_ble #532: Check if bluetooth is active, through flutter_reactive_ble

rafaelrpedrosa, 2022-03-11:
I need to check if bluetooth is active, before the scanner, I didn't find anything similar in flutter_reactive_ble.
Any suggestions on how to do this check?

sh3rlock97, 2022-03-11:
Follow the status-Stream
```

    return StreamBuilder<BleStatus>(
        stream: _ble.statusStream,
        builder: (context, AsyncSnapshot<BleStatus> snapshot) {
         Container(
              color: Theme.of(context).canvasColor,
              width: double.infinity,
              height: double.infinity,
              child: Padding(
                padding: const EdgeInsets.all(20.0),
                child: _buildStatusWidget(context, snapshot.data),
              ),
          );
        });
```

Taym95, 2022-03-12:
Use statusStream, please check example app

closed 2022-03-12 by Taym95