PhilipsHue/flutter_reactive_ble #535: Scan is empty on Android 12

mbabuglia, 2022-03-15:
**Describe the bug**
On Android 12 the scan results of the example app is empty, this worked fine on Android 11.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the example app for the first time. Widget BleStatusScreen shows unauthorized as expected.
2. On Android app settings, manually give "Nearby devices" permissions, return to the app.
3. Status correctly updates. Input the service id used by an available device and start scanning. You will see no device is returned.

The exactly same steps work fine on Android 10 and 11.
(albeit the permission given manually is Location instead of Nearby devices).

**Expected behavior**
The available device is discovered and shown in the UI.

**Smartphone / tablet**
 - Device: Samsung Galaxy S20 FE
 - OS: Android 12
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: [e.g. Fitbit Versa, iPhone SE, or CUSTOM]
- Does it run a custom firmware / software: yes / no

**Additional context**
The debug console shows no errors while/after scanning.

Flutter doctor:

> [√] Flutter (Channel stable, 2.10.3, on Microsoft Windows [Version 10.0.22000.556], locale en-US)
[√] Android toolchain - develop for Android devices (Android SDK version 31.0.0)
[X] Chrome - develop for the web (Cannot find Chrome executable at .\Google\Chrome\Application\chrome.exe)
    ! Cannot find Chrome. Try setting CHROME_EXECUTABLE to a Chrome executable.
[√] Visual Studio - develop for Windows (Visual Studio Build Tools 2019 16.11.9)
[√] Android Studio (version 2021.1)
[√] IntelliJ IDEA Ultimate Edition (version 2021.2)
[√] VS Code (version 1.65.2)
[√] Connected device (3 available)
[√] HTTP Host Availability

PieterAelse, 2022-03-16:
> (albeit the permission given manually is Location instead of Nearby devices).

Did you try with requesting the Nearby devices permission on Android 12? What's your `targetSdkVersion` in your `AndroidManifest.xml`?

mbabuglia, 2022-03-16:
Hello @PieterAelse . targetSdkVersion is 31 in build.gradle.
The app correctly asks for Nearby devices permission, is after giving this permission that the UI refreshes and allows me to start the scan.

I haven't touched the example app except to update gradle version from 7.0.2 to 7.1.0 
[This is the example ](https://github.com/PhilipsHue/flutter_reactive_ble/tree/master/example)I am talking about.

jdinh1, 2022-03-18:
@mbabuglia Are you still seeing the same thing? I'm also having issue scanning. It was working fine just a week ago for me, and nothing has changed in the code base since. Permissions are properly granted for both `Location` and `Nearby devices`. The error was 
```
Scan failed because application registration failed (code 6)
```
This is on Android 12 with `targetSdkVersion: 31`. I can confirm that it's still working on Android 11. I will try to look more into this once I have more time this weekends.

mbabuglia, 2022-03-18:
@jdinh1 , yes, still hapening for me. I can confirm it was working on Android 11 as well but I can't confirm it worked on Android 12, since this error happened to me as soon as I switched to Android 12.

I am not seeing the error you mention on my logs. In my case, the scan just doesn't find anything and doesn't throw any errors.

`I/BluetoothAdapter( 6097): STATE_ON
D/BluetoothLeScanner( 6097): Start Scan with callback
D/InsetsSourceConsumer( 6097): setRequestedVisible: visible=false, type=19, host=com.signify.hue.reactivebleexample/com.signify.hue.reactivebleexample.MainActivity, from=android.view.InsetsSourceConsumer.hide:242 android.view.ImeInsetsSourceConsumer.hide:68 android.view.ImeInsetsSourceConsumer.hide:74 android.view.InsetsController.hideDirectly:1472 android.view.InsetsController.controlAnimationUnchecked:1138 android.view.InsetsController.applyAnimation:1455 android.view.InsetsController.applyAnimation:1436 android.view.InsetsController.hide:1005 android.view.ViewRootImpl$ViewRootHandler.handleMessageImpl:6348 android.view.ViewRootImpl$ViewRootHandler.handleMessage:6269
D/BluetoothLeScanner( 6097): onScannerRegistered() - status=0 scannerId=16 mScannerId=0
I/IMM_LC  ( 6097): notifyImeHidden
`

jdinh1, 2022-03-18:
@mbabuglia We may have different issues then. For what it's worth, here's the snippet of code I'm using to request all the required permissions on Android 12. You can try to see if it works on your app instead of manually give it permissions. It uses the `permission_handler` package and should be called before `runApp()`.

```
Future<void> _checkPermissions() async {
  if (Platform.isAndroid) {
    Map<Permission, PermissionStatus> statuses = await [
      Permission.location,
      Permission.bluetoothScan,
      Permission.bluetoothConnect
    ].request();

    for (var status in statuses.entries) {
      if (status.key == Permission.location) {
        if (status.value.isGranted) {
          debugPrint('Location permission granted');
        } else {
          debugPrint("Location permission not granted");
        }
      } else if (status.key == Permission.bluetoothScan) {
        if (status.value.isGranted) {
          debugPrint('Bluetooth scan permission granted');
        } else {
          debugPrint('Bluetooth scan permission not granted');
        }
      } else if (status.key == Permission.bluetoothConnect) {
        if (status.value.isGranted) {
          debugPrint('Bluetooth connect permission granted');
        } else {
          debugPrint('Bluetooth connect permission not granted');
        }
      }
    }
  }
}
```

That should get all the required permissions to get example app to run on Android 12, but for whatever reason the recent security patch I let go through on my main Android 12 phone is wrecking everything up.

mbabuglia, 2022-03-18:
@jdinh1, I am using permission_handler as well in my real application but i am sticking to the library example as a "Minimum code to reproduce the issue", which doesn't use it.

One note regarding your permissions, location permission shouldn't be needed anymore if you declare 
`<uses-permission
            android:name="android.permission.BLUETOOTH_SCAN"
            android:usesPermissionFlags="neverForLocation" />`

in your AnddroidManifest.xml. Take a look at [the example's manifest](https://github.com/PhilipsHue/flutter_reactive_ble/blob/master/example/android/app/src/main/AndroidManifest.xml).

(BTW, I was able to connect to my device on Android 12 using SPP instead of GATT withouth declaring location permissions).

I realized now that devices ARE being found if I scan passing an empty services id list, but my device isn't there. Before was passing the specific service if of my device so I didn't see any results at all. Very odd since the same device is discoverd in Android 11.

I will continue investigating.

jdinh1, 2022-03-18:
@mbabuglia Thanks for the suggesting. I will try it out this weekend. 

Also, by chance, have you try scanning and connecting to your device using another app like `nrF Connect` to see if your device still show valid service id and characteristics? Very strange to see your app able to discover other devices but not yours.

Taym95, 2022-06-13:
Please follow README instruction and example app.

closed 2022-06-13 by Taym95