PhilipsHue/flutter_reactive_ble #536: OTA Update

albo1337, 2022-03-17:
Hello all!

I am trying to do an OTA Update on my bluetooth device. Therefor I got some gbl files, which I need to read out and send via flutter_reactive_ble to the bluetooth device. Does anyone of you know how to readout the gbl file and then convert it, that I can use writeCharacteristicWithResponse for this?

Thank you a lot!
Albo

yuriboeira11tx, 2022-03-17:
For OTA update, I'm using a .bin file, reading it and transforming it into a List<int> in sizes of 512, data packets that will be transferred in that size. Detail: you must configure the MTU size in both systems, application and firmware.

albo1337, 2022-03-17:
@yuriboeira11tx  thanks for your reply!
Can you show me how you are reading it out?