PhilipsHue/flutter_reactive_ble #537: Can't build on M1 Macbook, Protobuf issue

DotDebian, 2022-03-17:
**Describe the bug**
When running my small test app, build is going well until the app launches in debug. It then crashes with a stack trace about Protobuf.

**To Reproduce**
Steps to reproduce the behavior:
1. Get a Macbook M1
2. Build an app with a scanForDevices
3. Run on an iPhone in debug mode
**Expected behavior**
Not to crash.

- [x] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: iPhone 12 Pro Max
 - OS: iOS 15
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: Custom
- Does it run a custom firmware / software: yes

**Additional context**
Android Studio run logs:
```
Launching lib/main.dart on iPhone de Quentin in debug mode...
Automatically signing iOS for device deployment using specified development team in Xcode project: 8547YCHLVR
Running Xcode build...
Xcode build done.                                           29,3s
Installing and launching...
(lldb) dyld[6018]: Library not loaded: @rpath/Protobuf.framework/Protobuf
  Referenced from: /private/var/containers/Bundle/Application/35F93AE4-95C2-49F5-A81C-45082EAB72C8/Runner.app/Runner
  Reason: tried: '/usr/lib/swift/Protobuf.framework/Protobuf' (no such file), '/usr/lib/swift/Protobuf.framework/Protobuf' (no such file), '/private/var/containers/Bundle/Application/35F93AE4-95C2-49F5-A81C-45082EAB72C8/Runner.app/Frameworks/Protobuf.framework/Protobuf' (no such file), '/private/var/containers/Bundle/Application/35F93AE4-95C2-49F5-A81C-45082EAB72C8/Runner.app/Frameworks/Protobuf.framework/Protobuf' (no such file), '/private/var/containers/Bundle/Application/35F93AE4-95C2-49F5-A81C-45082EAB72C8/Runner.app/Frameworks/Protobuf.framework/Protobuf' (no such file), '/usr/lib/swift/Protobuf.framework/Protobuf' (no such file), '/usr/lib/swift/Protobuf.framework/Protobuf' (no such file), '/private/var/containers/Bundle/Application/35F93AE4-95C2-49F5-A81C-45082EAB72C8/Runner.app/Frameworks/Protobuf.framework/Protobuf' (no such file), '/private/var/containers/Bundle/Application/35F93AE4-95C2-49F5-A81C-45082EAB72C8/Runner.app/Frameworks/Protobuf.framework/Protobuf' (no such file), '/private/var/containers/Bundle/Application/35F93AE4-95C2-49F5-A81C-45082EAB72C8/Runner.app/Frameworks/Protobuf.framework/Protobuf' (no such file), '/System/Library/Frameworks/Protobuf.framework/Protobuf' (no such file)
* thread #1, stop reason = signal SIGABRT
    frame #0: 0x00000001043bc578 dyld`__abort_with_payload + 8
dyld`__abort_with_payload:
->  0x1043bc578 <+8>:  b.lo   0x1043bc598               ; <+40>
    0x1043bc57c <+12>: pacibsp 
    0x1043bc580 <+16>: stp    x29, x30, [sp, #-0x10]!
    0x1043bc584 <+20>: mov    x29, sp
Target 0: (Runner) stopped.
Error launching application on iPhone de Quentin.
```

Using Product -> Run on Xcode works tho (it crashed on the first try, but went OK on the second time).
This is kinda annoying as I cannot properly debug my app without Android Studio.

DotDebian, 2022-03-17:
Looks like `flutter clean` did the job. Closing this issue, will maybe help someone someday.

DotDebian, 2022-03-17:
After flutter clean, app is running smoothly but on any hot reload or setState, the following happens (with a crash):
```reactive_ble_mobile/PluginController.swift:158: Assertion failed
reactive_ble_mobile/PluginController.swift:158: Assertion failed
* thread #1, queue = 'com.apple.main-thread', stop reason = Assertion failed
    frame #0: 0x00000001852fcd7c libswiftCore.dylib`_swift_runtime_on_report
libswiftCore.dylib`_swift_runtime_on_report:
->  0x1852fcd7c <+0>: ret    
libswiftCore.dylib`_swift_reportToDebugger:
    0x1852fcd80 <+0>: b      0x1852fcd7c               ; _swift_runtime_on_report
libswiftCore.dylib`_swift_shouldReportFatalErrorsToDebugger:
    0x1852fcd84 <+0>: adrp   x8, 335449
    0x1852fcd88 <+4>: ldrb   w0, [x8, #0x2f4]
Target 0: (Runner) stopped.
Lost connection to device.
```

Taym95, 2022-03-17:
The title of the issue doesn't match the content of the issue, can you rename or create another issue with reproducing scenario?