PhilipsHue/flutter_reactive_ble #539: Reconnection to device not working with Android when turning dfu mode on

albo1337, 2022-03-23:
**Describe the bug**
When I turn the DFU-mode on a Hardware using Android, I am not able to find the hardware again. The reconnection is not working. On iOS this works, it seems that on Android it doesn't work as expected.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to one device using `connectTo()` 
2. Write a 0 into the dfu characteristic
3. Now hardware is doing a reboot
4. scan for devices
5. no devices can be found, even though the device can be found using an iOS phone

**Expected behavior**
Device can be found on Android and on iOS

- [ yes] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Pixel 4 XL
 - OS: Android 12
 - Package version: newest

albo1337, 2022-03-23:
This is what I get as soon as I turn DFU mode on:

`ConnectionStateUpdate(deviceId: 68:0A:E2:F1:88:B1, connectionState: DeviceConnectionState.disconnected, failure: GenericFailure<ConnectionError>(code: ConnectionError.failedToConnect, message: "Disconnected from MAC='XX:XX:XX:XX:XX:XX' with status 19 (GATT_CONN_TERMINATE_PEER_USER)"))`

jobfeikens, 2022-03-26:
Are you using NRF DFU? In that case when you use unbonded DFU, the last digit of the device's mac address increments with 1, so it will become B2. The recommended way however is to change the devices dfu advertising name using `DFU_OP_SET_ADV_NAME` and then scan for a device with that name because you can't see the mac address on iOS.

albo1337, 2022-03-30:
Hello @jobfeikens, I have a BGM13S Blue gecko chip in my hardware. so no NRF for me ....

rohitsangwan01, 2022-04-23:
@albo1337 in my case , i use this combo

set device to dfu
rename with a random string

now that device is advertising in dfu with that random name

and scan for bleDevices , get device with that random name out of it

now you got dfu Device ID , use that to update device

also make sure to store orignal device id , to reconnect after dfu operation