PhilipsHue/flutter_reactive_ble #541: Bluetooth Windows Support

dlewis2017, 2022-03-24:
**Is your feature request related to a problem? (please describe)**
Seeing as Flutter is now in production for Windows, are there plans for this plugin to be extended to Windows?

**Describe the solution you'd like**
I assume exposing/creating a Dart wrapper for the the [Win32 C++ bluetooth API's](https://docs.microsoft.com/en-us/windows/win32/api/_bltooth/) would suffice but I'm not as familiar with Flutter native plugin

**Describe alternatives you've considered**
The [C Interlop](https://docs.flutter.dev/development/platform-integration/c-interop) I guess could be used for those needing Windows support but at that point, might as well make it a feature of this library 

**Additional context**
Separate from this Linux issue: https://github.com/PhilipsHue/flutter_reactive_ble/issues/246

Taym95, 2022-03-25:
Hi @dlewis2017 , we are planning to support Windows but not inside this repository but as an official endorsed plugin, and forwarding windows issues to that specific repo, we do not have the capacity to do maintenance on other platforms, we want to keep IOS/Android on track

closed 2022-03-25 by Taym95

farr64, 2022-03-25:
Before you go: Would you please also consider an official endorsed plugin for macOS and an official endorsed plugin for the Web?

Thanks.

dlewis2017, 2022-03-29:
@Taym95 You wouldn't happen to know when that plugin will be pushed out in alpha mode would you? As well as where the project is being developed (for tracking).