PhilipsHue/flutter_reactive_ble #548: requestMtu returns zero after reconnect

hohlfma, 2022-04-11:
**Describe the bug**
Hi there! First of all, thanks for the great work of making this library! :-)

I have an issue with re-using an already existing connection after hot-restarting my app.

After app startup, I listen to the global `connectedDeviceStream` to find out if there is already an active connection to my device to just reconnect to it in this case (to reclaim the already open connection).

In case there is already a connection, I get the following state update from the `connectedDeviceStream`:

```
I/flutter (27660): ConnectionStateUpdate(deviceId: E5:D0:72:03:1F:86, connectionState: DeviceConnectionState.connected, failure: null)
```

To reclaim this connection, I now call `connectToDevice` with the device id from the state update, which works fine:

```
I/flutter (27660): BLE: Connect to device E5:D0:72:03:1F:86...
I/BluetoothAdapter(27660): STATE_ON
D/BluetoothGatt(27660): connect() - device: E5:D0:72:03:1F:86, auto: false
I/BluetoothAdapter(27660): isSecureModeEnabled
D/BluetoothGatt(27660): registerApp()
D/BluetoothGatt(27660): registerApp() - UUID=a03c16eb-4cc3-45c6-b10d-73c4c03c117f
D/BluetoothGatt(27660): onClientRegistered() - status=0 clientIf=13
D/BluetoothGatt(27660): onClientConnectionState() - status=0 clientIf=13 device=E5:D0:72:03:1F:86
I/flutter (27660): Connection state: ConnectionStateUpdate(deviceId: E5:D0:72:03:1F:86, connectionState: DeviceConnectionState.connecting, failure: null)
I/flutter (27660): Connection state: ConnectionStateUpdate(deviceId: E5:D0:72:03:1F:86, connectionState: DeviceConnectionState.connected, failure: null)
I/flutter (27660): Start discovering services for: E5:D0:72:03:1F:86
D/BluetoothGatt(27660): discoverServices() - device: E5:D0:72:03:1F:86
D/BluetoothGatt(27660): onSearchComplete() = Device=E5:D0:72:03:1F:86 Status=0
```

But if I now try to configure the MTU again (which already has been done when the connection has been opened initially), `requestMtu` is returning zero:

[image]

However, the logs show the correct value (MTU = 23 in this case), together with a `status=4` (which may be the cause of the issue?):

```
D/BluetoothGatt(27660): configureMTU() - device: E5:D0:72:03:1F:86 mtu: 23
D/BluetoothGatt(27660): onConfigureMTU() - Device=E5:D0:72:03:1F:86 mtu=23 status=4
```

**Expected behavior**
I know that MTU can't be changed after initial negotiation - that`s fine. But I would expect that requestMtu() just returns the actual MTU in this case, as this is the only way to ask for the negotiated MTU at the moment.

If `requestMtu` would just return the actual MTU of 23 in this case, everything would be fine and I could perfectly re-use the existing connection even after hot restart. Alternatively it would be okay when `requestMtu` fails in this case, if there would be another way to find out the actual MTU of a device, e.g. another function `getMtu(deviceId)` or something like that.

Thanks in advance for any hints and support on how to overcome this issue! 

**Smartphone / tablet**
 - Device: Samsung Galaxy Tab S6 Lite
 - OS: Android 11
 - Package version: 5.0.2

**Peripheral device**
- Custom device (based on Nordic nRF52)

dusalex, 2022-06-29:
I have experienced similar issues with a Samsung A03s. On repeated connections to the same device, requestMtu returns 0 after exactly 30 seconds (so I assume that's the result of some kind of timeout). I'm not sure if its related to requestMtu, or if this just happens to be the first BLE action we are doing after connecting.

In any case, it works fine on other Android smartphones, so maybe there is something particular about Samsung devices.

Taym95, 2022-10-24:
I dont think this is library issue, I will close it for now

closed 2022-10-24 by Taym95

hohlfma, 2022-10-24:
Hi @Taym95, thanks for the feedback!

Why do you think that this is not a library issue? It is a library function (requestMtu) that is called here and that returns a wrong MTU, even if is seems like the other layers know about the correct value:

D/BluetoothGatt(27660): onConfigureMTU() - Device=E5:D0:72:03:1F:86 mtu=23 status=4

As there is no other way to query for the current MTU and requestMTU also failes to return the current value, the user of the library has no clue about which MTU is actually used. What would you do in such a case?

Thanks and best regards

Taym95, 2022-10-24:
If it work for all devices and only Samsung A03s not woking do you think it is library issue?

dusalex, 2022-10-24:
Actually, there are reports for two different devices (Samsung Galaxy Tab S6 Lite, Samsung A03s) in this thread. Most likely there are more.

mzohren, 2022-11-04:
Same problem here, devices used are Samsung SX200 and SM T500 on Android12, connecting to Zebra ZD421.
First connect returns correct MTU, after reconnect requestMTU returns 0.
I fixed that by caching the MTU locally, but it's a workaround that I'm not very happy about ...