PhilipsHue/flutter_reactive_ble #551: iOS writeCharacteristicWithoutResponse gives characteristic not writable error

andrewdixon1000, 2022-04-13:
**Describe the bug**
I'm running the example app for the package.  I connect to my custom peripheral and select my (protected) writable characteristic.  I enter a value into the value text field and press "Without response" button.  I see the exception:

_Exception (Exception: GenericFailure<WriteCharacteristicFailure>(code: WriteCharacteristicFailure.unknown, message: "The characteristic 6E400002-B5A3-F393-E0A9-E50E24DCCA9E of the service 6E400001-B5A3-F393-E0A9-E50E24DCCA9E of the peripheral 4F19023B-EB8D-A5C6-CBFC-750B2E04A481 is not writable"))

Pressing the "With response" is successful, but I don't want to use this in my application as I'm sending a lot of data to my characteristic and with response adds a lot of overhead and roughly doubles my transmission time.

**To Reproduce**
Steps to reproduce the behavior:
1. Forget device in iOS bluetooth settings
2. Connect to device and discover services
3. Select characteristic and send value "with response".  This triggers iOS to show the "Bluetooth Pairing Request" dialog
4. Pair with the device. I now see Output: Ok
5. Send value again "with response". Again I see Output: Ok but no pairing dialog, indicating that I am successfully paired with the device
6. Send value again "without response".  I get the exception described above

**Expected behavior**
I expect to be able to send data using "writeCharacteristicWithoutResponse"

I can successfully communicate with the peripheral from the same iOS device with nRF connect and BLE Scanner apps

**Smartphone / tablet**
 - Device: iPhone SE
 - OS: iOS 13.7
 - Package version: 5.0.2

**Peripheral device**
- Custom

**Additional context**
Works flawlessly on Android, I'm only having a problem with iOS

adnanjpg, 2022-04-23:
Hi, I am having this issue too. as @andrewdixon1000 mentioned, the package is just flawless on Android, thank you for the great work you have done. 

here is my `flutter doctor` output:
```
[✓] Flutter (Channel stable, 2.10.5, on macOS 12.3.1 21E258 darwin-arm, locale en-TR)
    • Flutter version 2.10.5 at /Users/adnanfahed/flutter
    • Upstream repository https://github.com/flutter/flutter.git
    • Framework revision 5464c5bac7 (5 days ago), 2022-04-18 09:55:37 -0700
    • Engine revision 57d3bac3dd
    • Dart version 2.16.2
    • DevTools version 2.9.2

[✓] Android toolchain - develop for Android devices (Android SDK version 33.0.0-rc2)
    • Android SDK at /Users/adnanfahed/Library/Android/sdk
    • Platform android-31, build-tools 33.0.0-rc2
    • Java binary at: /Applications/Android Studio.app/Contents/jre/Contents/Home/bin/java
    • Java version OpenJDK Runtime Environment (build 11.0.11+0-b60-7772763)
    • All Android licenses accepted.

[✓] Xcode - develop for iOS and macOS (Xcode 13.3.1)
    • Xcode at /Applications/Xcode.app/Contents/Developer
    • CocoaPods version 1.11.3

[✓] Chrome - develop for the web
    • Chrome at /Applications/Google Chrome.app/Contents/MacOS/Google Chrome

[✓] Android Studio (version 2021.1)
    • Android Studio at /Applications/Android Studio.app/Contents
    • Flutter plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/9212-flutter
    • Dart plugin can be installed from:
      🔨 https://plugins.jetbrains.com/plugin/6351-dart
    • Java version OpenJDK Runtime Environment (build 11.0.11+0-b60-7772763)

[✓] VS Code (version 1.66.2)
    • VS Code at /Applications/Visual Studio Code.app/Contents
    • Flutter extension version 3.38.1

[✓] Connected device (2 available)
    • sdk gphone arm64 (mobile) • emulator-5554 • android-arm64  • Android 11 (API 30) (emulator)
    • Chrome (web)              • chrome        • web-javascript • Google Chrome 100.0.4896.127
    ! Error: Adnan’s iPhone X is not connected. Xcode will continue when Adnan’s iPhone X is connected. (code -13)

[✓] HTTP Host Availability
    • All required HTTP hosts are available

• No issues found!
```
my computer specifiactions:
```
System Version: macOS 12.3.1 (21E258)
Kernel Version: Darwin 21.4.0
Chip: Apple M1 Pro
```

my phone specifiactions:
```
Model Name: iPhone X
Software Version: iOS 15.4.1
```

akospwc, 2022-04-24:
Does your custom peripheral allow writes without response? On my STM32-based custom peripheral, writing with and without response are separate characteristic properties.

adnanjpg, 2022-04-24:
hi @akospwc , I am not having any problems on android, and in fact I've tested the write with response and without functions' performances and I've seen big difference, doesn't that indicate that my hardware is accepting writing without response? I am pretty new to BLE so my observations might be absurd

adnanjpg, 2022-04-24:
My peripheral is a `ESP32 WROOM 32D` with `blueetooth 4.2` and `arduino`

akospwc, 2022-04-24:
The difference is indeed pretty big because the BLE stack does not need to wait for a confirmation after every packet is sent to the peripheral. (However, packet loss is something you will need to prepare for, but that's a different problem)

Are you positive that the hardware is actually accepting the sent packets on Android? Is there a way for you to verify this on the receiving end?  Do you have access to the peripheral's source code to verify if the characteristic is set up for writing with and without response? If not, you can use the LightBlue app on your iOS/Android device to connect to the peripheral and list its characteristics. It will list all characteristics and also display its properties (Readable, Writable, Writable Without Response, etc.)

adnanjpg, 2022-04-24:
We have even built playing audio functionality that sends chunks of audio without response and that worked pretty well on android, and tried the same functionality with response, which gave us a very big difference. and also we've debugged on nrf connect and our characteristics show that they allow all 4 properties (read, write, indicate, notify) (I am not able to provide a screenshot right now, I'll add it later). so we're pretty sure that our peripheral accepts write without response

adnanjpg, 2022-04-25:
here is an nrf connect screenshot of a characteristic that works just fine on android on writing without response, and have thrown an error on iOS

[image]

adnanjpg, 2022-04-25:
actually this screenshot is from an Android NRF Connect. I will provide one from an iOS soon

adnanjpg, 2022-04-25:
i think this problem maybe releated to our hardware. we've started investigating, can you please also provide a screenshot of the overview of the device from the `devices screen` > `scanner tab`, which would look like this [image]

akospwc, 2022-04-25:
Here you go. 
[image]

andrewdixon1000, 2022-04-28:
Thanks @akospwc and @adnanjpg - your posts have led me to the cause of my problem.  My bluetooth device's characteristic was not configured to access write without response.  I still don't understand why it was working with Android despite this.  I've had the firmware developer update the configuration of the characteristic and I now have it working with Android and iOS.  Thanks again.

adnanjpg, 2022-05-06:
@akospwc sorry for the late reply, and thank you so much for assisting us through this so patiently! I've added a `WRITE NO RESPONSE` property to my device and now it works exactly as expected. again, thanks for your effort, I think we can close this issue now.

akospwc, 2022-05-06:
I'm glad I could help. @andrewdixon1000 , please feel free to close the ticket. (I'm not associated with the project, I'm just a fellow user of this excellent library)

closed 2022-05-10 by andrewdixon1000