PhilipsHue/flutter_reactive_ble #556: Reactive_BLE example not working,not connecting to device discovered

thusithadeepal9356, 2022-05-03:
**Describe the bug**
I used the example https://github.com/PhilipsHue/flutter_reactive_ble/tree/master/example/
**To Reproduce**
Steps to reproduce the behavior:
Discover devices MacBook pro and Huawei Y9 phone,but click connect, connecting appear for long time,not connecting to device

**Expected behavior**
Need to connect the device.

**Smartphone / tablet**
 - Device: [e.g.Huawei Y9 2018]
 - OS: [Macbook pro]
 - Package version: [flutter_reactive_ble: ^5.0.2]

RaiJanvi, 2022-05-18:
Try adding internet permission in Android Manifest file.
<uses-permission android:name="android.permission.INTERNET"/>

Taym95, 2022-06-13:
> Try adding internet permission in Android Manifest file. `<uses-permission android:name="android.permission.INTERNET"/>`

Yes, you can add  `<uses-permission android:name="android.permission.INTERNET"/>`, note: this is needed only in debug!

thusithadeepal9356, 2022-06-17:
@Taym95  @RaiJanvi  It is already added to android manifest