PhilipsHue/flutter_reactive_ble #558: If a new connection is initiated before the last connection timed out, the connection to the device cannot be started, and there is no time-out feedback

kexuebiao, 2022-05-12:
**Describe the bug**
Two bluetooth devices A and B, when the first device A start the ble connection, and 
Before A connection times out, start to connect device B, and disconnecting device A, device B cannot be connected, nor can it report an error after the timeout
```
import 'dart:async';

import 'package:flutter/material.dart';
import 'package:flutter_reactive_ble/flutter_reactive_ble.dart';

class BlePage extends StatefulWidget {
  const BlePage({Key? key}) : super(key: key);

  @override
  State<BlePage> createState() => _BlePageState();
}

class _BlePageState extends State<BlePage> {
  final flutterReactiveBle = FlutterReactiveBle();
  StreamSubscription? _connectionSubscription;

  @override
  void initState() {
    super.initState();

    flutterReactiveBle.logLevel = LogLevel.verbose;
    flutterReactiveBle.connectedDeviceStream.listen((event) {
      doLog(
          "connectedDeviceStream(event) ----> ${event.deviceId} ${event.connectionState}");
    });
  }

  Future<void> connect(String foundDeviceId) async {
    doLog("---- connectTo ---- $foundDeviceId");
    await _connectionSubscription?.cancel().whenComplete(() {
      _connectionSubscription = null;
    });

    _connectionSubscription = flutterReactiveBle
        .connectToDevice(
      id: foundDeviceId,
      connectionTimeout: const Duration(seconds: 15),
    )
        .listen((connectionState) {
      doLog("---- $connectionState");
    }, onError: (Object error) {
      doLog("---- onError ------------------------------");
      print(error);
    }, onDone: () {
      doLog("---- onDone --------------------------------");
    }, cancelOnError: true);
  }

  @override
  Widget build(BuildContext context) {
    return Scaffold(
      appBar: AppBar(
        title: const Text("ble connect"),
      ),
      body: Center(
        child: Column(
          crossAxisAlignment: CrossAxisAlignment.start,
          children: [
            ElevatedButton(
                onPressed: () async {
                  connect("18:3B:26:01:73:67");
                },
                child: const Text("connect to A")),
            ElevatedButton(
              onPressed: () async {
                connect("18:3B:26:01:73:68");
              },
              child: const Text('connect to B'),
            ),
          ],
        ),
      ),
    );
  }

  void doLog(String msg) {
    debugPrint("${DateTime.now()} $msg");
  }
}

```

```
Performing hot reload...
Syncing files to device Y3L...
Reloaded 1 of 1234 libraries in 1,570ms.
I/flutter ( 7227): 2022-05-13 03:10:35.210270 ---- connectTo ---- 18:3B:26:01:73:67
D/BluetoothGatt( 7227): connect() - device: 18:3B:26:01:73:67, auto: false
D/BluetoothGatt( 7227): registerApp()
D/BluetoothGatt( 7227): registerApp() - UUID=ca1279ec-ee3c-45f0-8342-88199033a90d
D/BluetoothGatt( 7227): onClientRegistered() - status=0 clientIf=5
I/flutter ( 7227): 2022-05-13 03:10:35.325659 connectedDeviceStream(event) ----> 18:3B:26:01:73:68 DeviceConnectionState.disconnected
I/flutter ( 7227): 2022-05-13 03:10:35.330939 connectedDeviceStream(event) ----> 18:3B:26:01:73:68 DeviceConnectionState.disconnected
I/flutter ( 7227): 2022-05-13 03:10:35.349389 ---- ConnectionStateUpdate(deviceId: 18:3B:26:01:73:67, connectionState: DeviceConnectionState.connecting, failure: null)
I/flutter ( 7227): 2022-05-13 03:10:35.351172 connectedDeviceStream(event) ----> 18:3B:26:01:73:67 DeviceConnectionState.connecting
I/flutter ( 7227): 2022-05-13 03:10:42.320905 ---- connectTo ---- 18:3B:26:01:73:68
D/BluetoothManager( 7227): getConnectionState()
D/BluetoothManager( 7227): getConnectedDevices
D/BluetoothGatt( 7227): close()
D/BluetoothGatt( 7227): unregisterApp() - mClientIf=5
I/flutter ( 7227): 2022-05-13 03:10:42.392600 connectedDeviceStream(event) ----> 18:3B:26:01:73:67 DeviceConnectionState.disconnected
I/flutter ( 7227): 2022-05-13 03:10:42.396749 connectedDeviceStream(event) ----> 18:3B:26:01:73:67 DeviceConnectionState.connecting
I/flutter ( 7227): 2022-05-13 03:10:42.403469 ---- ConnectionStateUpdate(deviceId: 18:3B:26:01:73:68, connectionState: DeviceConnectionState.connecting, failure: null)
I/flutter ( 7227): 2022-05-13 03:10:42.405064 connectedDeviceStream(event) ----> 18:3B:26:01:73:68 DeviceConnectionState.connecting

```

**To Reproduce**
Steps to reproduce the behavior:
1. Click the button "connect to A" to start connecting device A, the connection timeout time is 15s
2. After 7s, click the button "connect to B" to start connecting to device B.
3. Device B does not initiate a Bluetooth connection, and after the connection times out, there is no error.

**Expected behavior**
It is hoped that the connection of device B can be started normally after switching the connection to device B before device A times out.

**Smartphone / tablet**
 - Device: HUAWEI nova8
 - OS: Harmony OS 2.0.0
 - Package version: flutter_reactive_ble: ^5.0.2

Stachy83, 2022-05-24:
Hi!

Did you find any explanation ?
I'm facing the same issue.

sifourquier, 2023-01-17:
Hello
Same issue for me.

wsmzdyss, 2023-05-23:
Same issue for me.

vjgoutham, 2023-06-27:
Hi , Same issue for me also. Is there any work around for this issue. Please let us know.

voyagerv0v, 2025-04-02:
same issue