PhilipsHue/flutter_reactive_ble #560: Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions

clockwork189, 2022-05-17:
**Describe the bug**
Hello, I am building an app that is targeting SDK 31 on Android. I am using this plugin along side a location tracking plugin(flutter_background_location). The location tracking plugin automatically adds permissions for COARSE and FINE location, however it seems like when I try to publish a built aab file to the android app store. I get the error of: 
**Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions**

**To Reproduce**
Steps to reproduce the behavior:
1. Add another location tracking plugin such as flutter_background_location or geolocator, along side flutter_reactive_ble.
2. Run the app (it will run fine in debug mode).
3. Run flutter build appbundle
4. Upload app bundle to google play appstore.

**Expected behavior**
It should pass and successfully upload.

**Smartphone / tablet**
 - OS: Android
 - Package version: ^5.0.2

clockwork189, 2022-05-17:
I went into the "Flutter Plugins"  section of my project, found "reactive_ble_mobile", navigated to: android>src>main>AndroidManifest.xml
[image]
It seems like maxSdkVersion=30 is set for both COARSE and FINE location. 

These are my Manifest permissions, as you can see there is no maxSdkVersion on the fine and coarse location permissions.
[image]

I tried removing the FINE and COARSE permissions from my Manifest entirely, however the permissions from "reactive_ble_mobile" will clash with other location libraries.

I ended up removing the "maxSdkVersion=30" declaration in Flutter Plugins>reactive_ble_mobile>android>src>main>AndroidManifest.xml then built my app and uploaded it to the playstore and this time I did not get a "Duplicate declarations of permission android.permission.ACCESS_COARSE_LOCATION with different maxSdkVersions" error, which is how I pin pointed this issue to this library. My current workaround works but I am afraid that it will get reset the next time i do a flutter clean + flutter pub get.

rickcasson, 2022-05-17:
It looks like this issue was fixed on the master branch, but was never published...
Try pointing the plugin directly to the repo master branch and see if that fixes it:

```
  flutter_reactive_ble:
    git:
      url: git@github.com:PhilipsHue/flutter_reactive_ble.git
      path: packages/flutter_reactive_ble
      ref: master
```
You will also need to override the package dependencies:

```
dependency_overrides:
  reactive_ble_mobile:
    git:
      url: git@github.com:PhilipsHue/flutter_reactive_ble.git
      path: packages/reactive_ble_mobile
      ref: master
  reactive_ble_platform_interface:
    git:
      url: git@github.com:PhilipsHue/flutter_reactive_ble.git
      path: packages/reactive_ble_platform_interface
      ref: master
```

rickcasson, 2022-05-17:
It might worth noting that this plugin depends on [RxAndroidBle](https://github.com/dariuszseweryn/RxAndroidBle) and will thus inherit all [AndroidManifest permissions from that plugin ](https://github.com/dariuszseweryn/RxAndroidBle/blob/master/rxandroidble/src/main/AndroidManifest.xml).

To prevent conflicts with you own permissions, in your AndroidManifest.xml file include:
```
  <uses-permission-sdk-23 android:name="android.permission.ACCESS_FINE_LOCATION" tools:node="remove"/>
  <uses-permission android:name="android.permission.ACCESS_FINE_LOCATION"/>
  <uses-permission-sdk-23 android:name="android.permission.ACCESS_COARSE_LOCATION" tools:node="remove"/>
  <uses-permission android:name="android.permission.ACCESS_COARSE_LOCATION"/>
```
You will also need to add `xmlns:tools="http://schemas.android.com/tools"` in the `manifest` tag at the top of the file.

VictorDutca, 2022-06-07:
Hi @rickcasson 
Actually the last proposed solution is not working.
In my use cae, the app needs localization permissions and if removed from manifest, for OS android 12 the authorization menu option will no longer appear in settings.

rickcasson, 2022-06-07:
@VictorDutca, could you share your AndroidManifest.xml file?

VictorDutca, 2022-06-08:
@rickcasson 
proposed solution:
[image]

before update:
[image]

VictorDutca, 2022-06-09:
@rickcasson 
The solution works, sorry for bothering. We had to change some CI parameters for the deploy and it worked.
Thank you

clockwork189, 2022-06-10:
Agreed, @rickcasson thanks for your solution!

Lootwig, 2022-06-29:
Is there a reason this issue was closed when the fix is not yet available as a stable release?

At the moment, the plugin can only be used with a direct git dependency, which seems more like a workaround than an actual fix (due to it breaking features like semantic versioning).

VictorDutca, 2022-06-29:
> It might worth noting that this plugin depends on [RxAndroidBle](https://github.com/dariuszseweryn/RxAndroidBle) and will thus inherit all [AndroidManifest permissions from that plugin [...]

This solution is working fine

vanlooverenkoen, 2023-05-09:
The documentation & comments in this issue did not work for me: 

This is what actually worked for me!
```
    <uses-permission
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="replace" />
    <uses-permission
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="replace" />

    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_COARSE_LOCATION"
        tools:node="remove" />
    <uses-permission-sdk-23
        android:name="android.permission.ACCESS_FINE_LOCATION"
        tools:node="remove" />
```