PhilipsHue/flutter_reactive_ble #561: Issue with subscribeToCharacteristic on iOS

aluu4445, 2022-05-17:
**Describe the bug**
If you subscribe to a characteristic, cancel the subscription, and then subscribe to the characteristic again, the second subscription doesn't emit events. 

**To Reproduce**
Steps to reproduce the behavior:
1. Subscribe to a characteristic using `subscribeToCharacteristic`
2. Write to the characteristic using `writeCharacteristicWithResponse` and observe events from the subscription from Step 1
3. Cancel the above subscription from Step 1
4. Subscribe to the same characteristic using `subscribeToCharacteristic`
5. Write to the characteristic using `writeCharacteristicWithResponse`
6. Observe that no events are emitted from the subscription from Step 4

**Expected behavior**
Expected behaviour is for the second subscription to emit events properly as the first subscription does.

**Smartphone / tablet**
 - Device: iPhone 8, iPhone 12
 - OS: 15.4.1
 - Package version: 5.0.2

**Additional context**
- If you add a short delay (300ms+) between cancelling the first subscription and re-subscribing, the second subscription emits events properly, but I would like to resolve this without adding a delay.
- This bug did not occur when I tested on Android 11 and Android 12 devices.

EdoardoVignati, 2022-05-20:
I can confirm this behavior. Actually if you try the example code, this problem doesn't happen. Cannot find out what's the real difference.

Taym95, 2022-06-13:
Please follow documentations and check example app.

closed 2022-06-13 by Taym95