PhilipsHue/flutter_reactive_ble #562: Version solving failed.

nbsathawara, 2022-05-19:
Hi guys,

I am using Restio (https://pub.dev/packages/restio) package in my existing flutter application to communicate with devices using Wifi Connection.

Now I have new devices with Bluetooth connectivity which I want to integrate in existing application.

So, when I am using flutter_reactiv_ble package, I am getting error of Version solving failed.

Here is my full error;

`Because flutter_reactive_ble >=5.0.2 depends on reactive_ble_mobile ^5.0.2 which depends on protobuf ^2.0.0, flutter_reactive_ble >=5.0.2 requires protobuf ^2.0.0.
And because protobuf >=2.0.0 depends on fixnum ^1.0.0, flutter_reactive_ble >=5.0.2 requires fixnum ^1.0.0.
And because raw 0.2.0 depends on fixnum ^0.10.0 and no versions of raw match >0.2.0 <0.3.0, flutter_reactive_ble >=5.0.2 is incompatible with raw ^0.2.0.
And because restio 0.10.11 depends on raw ^0.2.0 and no versions of restio match >0.10.11 <0.11.0, flutter_reactive_ble >=5.0.2 is incompatible with restio ^0.10.11.
So, because osdemo depends on both restio ^0.10.11 and flutter_reactive_ble ^5.0.2, version solving failed.
pub get failed (1; So, because osdemo depends on both restio ^0.10.11 and flutter_reactive_ble ^5.0.2, version solving failed.)
exit code 1`

Has any one of you faced this issue?
 Any help would be appreciated.

Thanks,
Nikhil.

Henge9, 2022-05-19:
This looks like a problem for restio, it is very outdated. But a quick fix can be to override the dependencies that are making the issue.
An example of dependency overrides(it is not a solution just a ugly fast fix):
```
dependency_overrides:
  fixnum: ^1.0.0
  ...
```

nbsathawara, 2022-05-24:
Found alternate package of restio.
Thanks for all the help.

closed 2022-05-24 by nbsathawara