PhilipsHue/flutter_reactive_ble #566: BLE frames are not detected on Android 12

Przemo1418, 2022-05-25:
**Describe the bug**
The library detects the peripheral devices general device MAC which you can use to connect, but no BLE frames. As we noticed the issue apeared after updating Android from 11 to 12 on the Samsung S10. Works perfectly on Samsung A21S [Android 11], Huawei P30 Pro [Android 11], POCO F3 [Android 11], iOS devices.

**To Reproduce**
Steps to reproduce the behavior:
Using Samsung S10/S20 Plus with Android 12
1. Detect devices using
flutterReactiveBle.scanForDevices(
withServices: List.empty(), scanMode: ScanMode.lowLatency)
.listen((device){
functionToCheckIfIsOurBeacon(device)
}
2. 'Device' object does not include BLE frames (BLE services are not avaliable, though you can connect to the device using GATT and change the devices parameters)

**Expected behavior**
Should be able to detect BLE frames/services

- [ ] I tried doing the same with a general BLE scanner application (e.g. [nRF Connect](https://duckduckgo.com/?q=nRF+connect)) and it exhibits the expected behavior as described above

**Smartphone / tablet**
 - Device: Samsung S10, Samsung S20 Plus
 - OS: Android 12
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: Estimote Proximity Beacon
- Does it run a custom firmware / software: no

**Additional context**
Issue is probably related to:
https://issuetracker.google.com/issues/228984309?pli=1

Screenshots of the 2 views representing the Bluetooth devices avaliable to connect and BLE frame view. On Samsung S10/S20Plus with Android 12 the BLE frame view ('Device' object) stays empty.
[image]
[image]

Airyzone, 2022-06-01:
I have the same problem, too.
For Android 12, iBeacon devices will not been detected. However, it works fine in Android 8 phone.

mbabuglia, 2022-06-10:
If your Bluetooth device is a Beacon, it is probably being filtered out by Android if you are using the new permissions of Android 12:

<uses-permission android:name="android.permission.BLUETOOTH_SCAN" usesPermissionFlags="neverForLocation" />

Check [my answer on StackOverflow](https://stackoverflow.com/a/72370969/666228).

Taym95, 2022-06-13:
@Przemo1418 can you please try this solution above?

closed 2022-06-13 by Taym95

Przemo1418, 2022-06-14:
Unfortunately that solution doesn't work, we couldn't get the actual permissions for bluetooth to detect the beacons using that solution.

AymenXD, 2022-07-08:
> Unfortunately that solution doesn't work, we couldn't get the actual permissions for bluetooth to detect the beacons using that solution. There was no popup nor any options in Android app settings to turn bluetooth permissions on manually.

is there a fix yet ? encountering the same problem . Can't detect any beacon !

bufolab, 2022-07-12:
Some Android 12 are not able to detect (even using native SDK) some beacons.
One Xiamoi phone was able to see some beacons. But Pixel3 or Samsung Note 20+ did not get any beacon but could get other BLE devices. All running Android 12.

Looks like some phones are filtering these beacons.

Przemo1418, 2022-07-12:
@Taym95 @AymenXD @bufolab 

**We managed to fix it by adding following to the manifest file:**

```
<manifest xmlns:android="http://schemas.android.com/apk/res/android"
    package="com.example.ukw_navigator"
    xmlns:tools="http://schemas.android.com/tools">
```

**and after that. lower in the file:**

```
<uses-permission android:name="android.permission.BLUETOOTH_SCAN" 
tools:remove="android:usesPermissionFlags"/>
```

AymenXD, 2022-07-12:
Thank You so much !!!! it finally worked  I’ve been searching for a solution during the whole past month :’)  From: Przemo1418Sent: 12 July 2022 13:52To: PhilipsHue/flutter_reactive_bleCc: AymenXD; MentionSubject: Re: [PhilipsHue/flutter_reactive_ble] [Beacons] BLE frames are not detected on Android 12 (Issue ***@***.*** @AymenXD @bufolabWe managed to fix it by adding following to the manifest file:<manifest xmlns:android="http://schemas.android.com/apk/res/android"    package="com.example.ukw_navigator"    xmlns:tools="http://schemas.android.com/tools">and after that. lower in the file:<uses-permission android:name="android.permission.BLUETOOTH_SCAN" tools:remove="android:usesPermissionFlags"/>