PhilipsHue/flutter_reactive_ble #567: Shouldn't this use Stream<Uint8List> for characteristic data?

Hackmodford, 2022-05-26:
**Is your feature request related to a problem? (please describe)**

Subscribing to a characteristic returns a stream of `List<int>`.

**Describe the solution you'd like**
Subscribing to a characteristic returns a stream of `Uint8List`.

**Describe alternatives you've considered**
Converting the data stream to `Uint8List` using `Uint8List.fromList()`

**Additional context**
This seems inefficient for processing byte streams. Shouldn't the stream be bytes instead of 64bit integers?

Taym95, 2022-06-13:
repo is just a ble package, we decided to let client convert `List<int>`

closed 2022-06-13 by Taym95

Hackmodford, 2022-06-14:
@Taym95 I'm afraid I don't understand. Could you please provide some more context?

Isn't it incredibly inefficient for 1 byte to be handled as 64 bit integers?