PhilipsHue/flutter_reactive_ble #568: iOS: background scan

msd05keisuke, 2022-05-27:
Hello.

I would like to do a BLE scan on an iOS device in background state.

But I can't do it at the moment. Is there any solution?

I am assuming the following case.

1, iBeacon region in
2, state suspend -> background (10sec)
3, connect to sensor during 10sec and upload sensed information to cloud

By doing this, I would like to achieve sensor-driven sensing, not user-driven.

info.plist
~~~
<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
	</array>
~~~

akospwc, 2022-05-31:
This should work without introducing iBeacons: your app initiates scanning for nearby devices using FlutterReactiveBle.scanForDevices(), and iOS automatically resumes your app to deliver the related events. After this, if your app connects to the discovered nearby devices, it will be able to receive incoming BLE packets. No 10-second timeout applies here.

Now, this only works until your user decides to swipe-kill the iOS app. When that happens, your app won't receive any BLE events until the user manually restarts the app. This is where iBeacons become useful, as iBeacon location events are delivered to terminated apps as well, and iOS will restart your app after swipe-kills or even after the phone is rebooted.  iBeacon functionality is not included in this library though, you'll need a separate package for that. We are using flutter_beacon for this.

msd05keisuke, 2022-06-01:
@akospwc 

I'm coding like this, but I can't scan in the background. Can you tell me the solution? I'm not good at English, so I'm sorry if there is a part that I can't communicate well.

~~~
// monitoring(use flutter_beacon)
    StreamSubscription streamMonitoring =
        flutterBeacon.monitoring(regions).listen((result) async {
      if (result.region.proximityUUID!.toLowerCase() ==
              BEACON_UUID.toLowerCase() &&
          result.monitoringState == MonitoringState.inside) {
        debugPrint("IN");

    // scan ble(use flutter_reactive_ble)
        flutterReactiveBle.scanForDevices(withServices: []).listen((device) {
        //code for handling results
        print(device.id + device.name);
        });

        
      } else if (result.monitoringState == MonitoringState.outside ||
          result.monitoringState == MonitoringState.unknown) {
        debugPrint("OUT");
      }
    });
~~~

closed 2022-10-24 by Taym95

HugoHeneault, 2023-10-03:
Hello @Taym95 

I'm also encoutering this kind of issue.

Here is my dart provider: 

```dart

class BleScanDevicesNotifier extends StateNotifier<List<DiscoveredDevice>> {
  final Ref ref;
  StreamSubscription<DiscoveredDevice>? subscription;

  BleScanDevicesNotifier(this.ref) : super([]) {
    Timer.periodic(const Duration(seconds: 1), (timer) {
      debugPrint('tick ${timer.tick.toString()}');
    });

    init();
  }

  init() {
    print('init');

    subscription?.cancel();
    subscription =
        ref.read(bleProvider).scanForDevices(withServices: []).listen((device) {
      final knownDeviceIndex = state.indexWhere((d) => d.id == device.id);

      print('received device ${device.id}');

      if (device.id == 'BFB9619B-89C0-D2E4-1574-B44D4EEE1455') {
        print(device);
      }

      if (knownDeviceIndex >= 0) {
        state[knownDeviceIndex] = device;
      } else {
        state.add(device);
      }

      state = [...state];
    }, onError: (err) {
      print(err);
      throw err;
    });
  }
}

final bleScanDevicesProvider =
    StateNotifierProvider<BleScanDevicesNotifier, List<DiscoveredDevice>>(
        (ref) => BleScanDevicesNotifier(ref));
```

When I start the app I have many `received device ... ` logs
and `tick 1`, 'tick 2`... logs

As soon as I switch to another app, without killing it, `received device...` logs stop, but I still have `tick 10`, `tick 11` coming (to ensure my provider keeps running in background)

I also tried to detect when app goes in background a call `init()` again; no luck. 

As soon as the app comes back to foreground, `received device...` logs come back again. 

Here is my `UIBackgroundModes`:

```
<key>UIBackgroundModes</key>
		<array>
			<string>fetch</string>
			<string>location</string>
			<string>bluetooth-central</string>
			<string>bluetooth-peripheral</string>
		</array>
```

Any ideas? 

Thanks for your help :)

HugoHeneault, 2023-10-03:
I also reproduce it simply on the example project : 

1. Start scanning
2. Enable verbose logging and check if XDebug console logs correctly
3. Send app to background : logs stops
Thanks for your help :)

HugoHeneault, 2023-10-03:
@msd05keisuke Did you find a way to get it working since 2022 ? 😇