PhilipsHue/flutter_reactive_ble #569: iOS has no name on discovered device on initial scan

robin-glimp, 2022-06-03:
**Describe the bug**
I tried to scan for my specific device on iOS but on initial scan the name is an empty string.
On a second scan the name appeared correctly.

**To Reproduce**
Steps to reproduce the behavior:
1. Start my custom device that was never discovered by the iPhone used for testing.
2. Start a scan using
```
_scanStream = _flutterReactiveBle.scanForDevices(
      withServices: [xxx],
      scanMode: ScanMode.lowLatency,
      requireLocationServicesEnabled: false,
    ).listen((device) {
      // only add if not in list already
      if (_foundDevices.every((element) => element.id != device.id)) {
        _foundDevices.add(device);
       // sort list alphabetically
        _foundDevices.sort((a, b) => a.name.compareTo(b.name));
      }
    });
```
3. In the `_foundDevices` list the devices show up, but with an empty string as the name
4. If I cancel the stream and then re-start it, the name shows up.
5. Also any subsequent scan shows the name

**Expected behavior**
I was hoping the name would pop up instantly when it was discovered. Otherwise this will be a very nasty hack in the code to start a rescan.

**Smartphone / tablet**
 - Device: all iOS devices
 - OS: iOS
 - Package version: 5.0.2

**Peripheral device**
- Vendor, model: CUSTOM
- Does it run a custom firmware / software: yes

Taym95, 2022-06-13:
This looks like device specific issue(custom firmware ), we can not do much about since we don't have your(custom firmware), did you test it in `nRF Connect app`? if not please do!

rintaro-deguchi, 2022-06-14:
same issue