PhilipsHue/flutter_reactive_ble #571: Can't scan for peripherals using extended advertising

BikeBr0, 2022-06-07:
**Describe the bug**
Since our peripheral implemented extended advertising I have not been able to find the peripheral when performing a scan. It was previously working when the peripheral used standard advertising. 

**To Reproduce**
Steps to reproduce the behavior:
1. Scan for peripherals using extended advertising  `_ble.scanForDevices(withServices: withServices)`

**Expected behavior**
To find the peripheral

**Smartphone / tablet**
 - Google Pixel 6 (Android ) 
 - Google Nexus 6P (Android 8.1.0)
 - flutter_reactive_ble: ^5.0.2

**Peripheral device**
- nRF52840-DK
- Does it run a custom firmware / software: yes

**Additional context**
This is an android specific bug as both advertising models work on iOS.

BikeBr0, 2022-06-08:
Further to the information provided above, we have tried using sample firmware provided with the nRF52840-DK and configured it to use Bluetooth 5 Advertising Extension. The peripheral shows up on nRF Connect on the Google Pixel 6 but not in my custom Flutter App. Is this not supported by flutter_reactive_ble?

rift-labs-developer, 2022-06-23:
Im having the same problem. It seems RxAndroidBle does not support extended advertising yet (https://github.com/dariuszseweryn/RxAndroidBle/issues/774 )

danielstuart14, 2022-07-27:
https://github.com/PhilipsHue/flutter_reactive_ble/pull/594

closed 2022-07-28 by Taym95

danielstuart14, 2022-08-02:
@Taym95 is there any ETA for a release with the fix?

Taym95, 2022-08-02:
@danielstuart14, I will make new release

danielstuart14, 2022-08-26:
Any news about that?