PhilipsHue/flutter_reactive_ble #575: After disconnecting  ble device  in android by cancelling the subscription the device is not showing in the respective  next scan .Need to kill the app and restart

vivek-regalrex, 2022-06-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Connect to device using connect method  connectToDevice(
2. disconnect using cancelling the connectToDevice subscription  .cancel()  method
3. doing a scan for devices using scanForDevices (behaviour is before starting new scan we will cancel the existing subscription)

**Expected behavior**
Device should show in next scan 

- [ ] NRF connect works fine 
Smartphone
 - Device: [MiA2]
 - OS: [Android 10]
 

**Peripheral device**
- Vendor, model: [e.g. Motor Pump]
- Does it run a custom firmware / software: yes 

**Additional context**
Add any other context about the problem here.

vivek-regalrex, 2022-06-20:
Please find the logs.somehow im seeing autoconnect is getting true after disconnecting the device might causing the device not to show in next scan and im providing connection timeout parameter

I/flutter (29755): disconnecting to device: 84:71:27:AA:62:D0
I/flutter (29755): disconnecting to device: 84:71:27:AA:62:D0

D/BluetoothGatt(29755): cancelOpen() - device: 84:71:27:AA:62:D0
D/BluetoothGatt(29755): onClientConnectionState() - status=0 clientIf=8 device=84:71:27:AA:62:D0
D/BluetoothGatt(29755): close()
D/BluetoothGatt(29755): unregisterApp() - mClientIf=8
D/BluetoothGatt(29755): connect() - device: 84:71:27:AA:62:D0, auto: true
D/BluetoothGatt(29755): registerApp()

Taym95, 2022-10-24:
> I had a similar problem, but with iOS devices. After connecting to the device, try to close the subscription to search for other devices, and when you disconnect from the device, start it again. Also, when disconnecting from the device, add a small delay before unsubscribing, this also helps to avoid some issues.
> 
> Example
> 
> ```
> await characteristicStream?.cancel();
> await Future.delayed(const Duration(milliseconds: 300));
> await deviceConnectionStateStream?.cancel();
> ```

This should fix your issue, I don't think this is library issues but rather an implementation issue.

closed 2022-10-24 by Taym95