PhilipsHue/flutter_reactive_ble #576: get a device's connection status

NewToProgramming58, 2022-06-17:
Hello, this isnt a bug. I am quite new to flutter. I would just like to know how to get the connection status of a specific device (connecting, connected, etc.). I have a list of devices that show on the scan and I can connect to a device but I cant be able to find a way to show the status in app. I have gon through the example but that looks like greek to me.

samnbd, 2022-06-19:
After first connecting to a device, you should store that device id in a shared preferences for example and use this method to listen to device connection stream:
`flutterReactiveBle.connectedDeviceStream.listen((state){
if(state.deviceId == your_device_id){
switch(state){
  case DeviceConnectionState.connected:
   //do something for connected state
  break;
 case DeviceConnectionState.disconnected:
  //do something for disconnected state
 break;
 case DeviceConnectionState.connecting:
  //do something for connecting state
 break;
}
}
});

closed 2022-06-21 by Taym95

NewToProgramming58, 2022-06-21:
> After first connecting to a device, you should store that device id in a shared preferences for example and use this method to listen to device connection stream:
>
> ``` [...]

Awesome thx got it to work <3