PhilipsHue/flutter_reactive_ble #578: Is there a way to use Bluetooth State Restoration on iOS?

samnbd, 2022-06-18:
So basically, I want to integrate subscribe to bluetooth events notifications sent by a wearable device even if the Flutter app is in a terminated state.
This can be done with background services on Android...but for iOS there is no such thing as maintaining bluetooth connection between iOS app and connected bluetooth device when the app is terminated.
iOS has what we call Bluetooth State Restoration that permits iOS to relaunch the app in the background when it receives some kind of message from a connected bluetooth device.
So how can we achieve this iOS process with this package?

closed 2022-06-20 by samnbd

abdaltintop, 2023-04-27:
Have you found a solution to this? @samnbd @dballance

dballance, 2023-04-27:
@abdaltintop - There is no solution short of forking and updating to add this. 

I've done that and will either contribute a PR back or add the fork here when it is ready.

You can't really just add restoration support to the `CBCentralManager` - you have to change some expectations of the library. Because of this - I'm unsure that `flutter_reactive_ble` will take these changes. 

For example, with a restored device you will not have a `StreamSubscription` you can cancel to "disconnect" the device. You'll have to have another method to directly disconnect a device.

Most of the method calls and whatnot are there - but you have to expose them. The whole "We just use streams and subscriptions" breaks down when you include state restoration. You have to have these additional methods exposed to properly handle restored devices.

NajdatAlAkkad, 2023-06-29:
@dballance are you still considering to create a PR or fork? I understand your response but I still need more technical information on how to achieve it.